MrChromebox/firmware #615: iPXE support

bam80, 2024-04-19:
It seems some work has been done but still the iPXE functionality was not included in the ROMs:
https://github.com/MrChromebox/firmware/issues/293#issuecomment-1551531786

Could we make it happen?

My fw:
>   ** Device: Asus Chromebox 3 / CN65 (TEEMO)
>     ** Fw Ver: MrChromebox-4.22.2-5-g822782a1001 (02/13/2024)

MrChromebox, 2024-04-19:
it's buggy so I disabled it, feel free to build your own with it enabled

bam80, 2024-04-19:
I'm afraid I lost experience to build my own already.

Do you remember what was buggy?
Any hope to solve it? Maybe you could build test fw for me?

The feature itself seems highly anticipated for me, even if it's somewhat buggy.

MrChromebox, 2024-04-19:
it only worked with WiFi, not the internal NIC. And caused some issues with USB detection somehow. It's not something I have time to debug currently

bam80, 2024-04-19:
Wow, I didn't expect it can work with WiFi!
This makes it even more attractive.

Those USB detection issues, were they persisted even if the feature was disabled in FW Setup?

MrChromebox, 2024-04-20:
disabled in FW setup? there was no runtime option to do so

bam80, 2024-04-20:
OK, I thought it's similar to Secure Boot which apparently has an option to be disabled in the FW Setup.

bam80, 2024-04-27:
> it only worked with WiFi, not the internal NIC.

Hmm, did it really work with Chrome devises WiFi?

I suppose all of them are equipped with Intel WiFi+BT adapters, which apparently have no iPXE drivers support:
https://ipxe.org/appnote/hardware_drivers

MrChromebox, 2024-04-27:
not all of them have Intel WiFi adapters

bam80, 2024-04-27:
What are the working in iPXE WiFi adapters they have then?
Are they Ath9k based?

To my understanding, most modern WiFi  adapters require blobs which prevents using them in iPXE.

MrChromebox, 2024-04-27:
I honestly have no recollection, coolstar was working on this, using 9elements' patches, and tested it / had it working. I merged the patches into my branch but had some issues with it so ended up not enabling it. Try asking on the chrultrabook forums, maybe coolstar or someone else there who tested it can provide more detail

bam80, 2024-04-28:
@coolstar could you give us some insight, please?

MrChromebox, 2025-11-22:
@walterav1984 I'm really trying to avoid adding any compiled binaries into my edk2 fork, especially when alternatives exist. The compiled drivers for those USB adapters add quite a bit of size to the final compressed payload as well.

iPXE, being fully open source, doesn't support any adapters which require any binary blobs (like Intel WiFi adapters). I don't know offhand if those USB adapters could easily be supported by porting the Linux drivers over. I tried to add support for the Realtek 2.5GBe PCIe NICs  but wasn't able to get them working in the few hours I spent on it.

bam80, 2025-11-22:
> iPXE has been added to the tree and will be in the next release

@MrChromebox Cool! Is it with WiFi support you mentioned earlier?

walterav1984, 2025-11-23:
It was indeed a quick fix to restore your previous included USB binary blobs to demonstrate iPXE additionally boots from USB network adapters, but I can understand that having those blobs in is not the way to go for the future codebase.

Going the CDC-NCM and CDC-ECM route is probably the right opensource solution, either native in EDK2 or with adjusting iPXE build options... still have to figure out either of them.

EDK2:
It seems your EDK2 fork already has [these drivers](https://github.com/tianocore/edk2/pull/4635/files) in place but I don't know how to activate them with build flags / kconfig options as in `DEFINE NETWORK_USB_ETHERNET = TRUE` . It also seems  some USB adapters can support 3 Drivers but need a [EDK2 USB quirk/workaround](https://github.com/tianocore/edk2/pull/6526) so it only expose the correct id to make CDC-NCM driver work. In theory people might even iPXE boot from their tethered Android phones if CDC-NCM is working or even RNDIS.

iPXE:
Read some things about building with `axge--ncm--ecm` but still have to figure out where to adjust these build options while including in EDK2.