MrChromebox/firmware #618: Suspend mode for OMNIGUL does not turn power indicator into blinking state

micmalti, 2024-04-22:
In suspend mode (`s2idle`), the power indicator light does not respond accordingly by turning amber orange in a blinking state, but staying as if the machine is still active. According to the systemd logs, the device is successfully entering suspend mode and it's not leaving it earlier than expected.

Such behaviour was noted on Ubuntu 23.10 and on the latest rolling release of both Manjaro (with kernel 6.6 and 6.8) and EndeavourOS. Interestingly, it was even present in ChromeOS, a sign that the problem might potentially be hardware-related.

A recent [pull request](https://github.com/MrChromebox/coreboot/pull/15) pointed towards the culprit being legacy timers breaking the s0ix state of Brya boards like OMNIGUL. Yet, the issue persisted after applying this fix.

MrChromebox, 2024-04-22:
the fact that the LED doesn't blink during S0ix under the stock firmware and ChromeOS indicates that's the way the EC firmware is programmed. Why do you think this is an actual problem?

micmalti, 2024-04-22:
Apparently, it's a feature, not a bug.

Maybe with `ectool`, I might be able to to set the indicator light into a blinking state during S0ix. Seeing that this isn't an issue anymore but a config preference, I'll close the issue.

closed 2024-04-22 by micmalti