MrChromebox/firmware #620: Morphius fullrom sleep/boot battery problem

bbyeen, 2024-04-23:
I'm running Ryzen 5 3500c 8GB 500GB NVME coreboot_edk2-morphius-mrchromebox_20240415.rom
I've noticed a thing where I can't boot first try into linux on battery (I've added iommu=pt as people suggested) and when waking from sleep it would freeze into black screen.
However when it's AC plugged it would always boot and wake up from sleep without issues.

MrChromebox, 2024-04-23:
running PopOS here on mine, no issues booting on battery or with suspend/resume.

What distro/kernel are you using?

bbyeen, 2024-04-23:
NixOS 6.8.7

MrChromebox, 2024-04-23:
best guess:  it might be an issue with your NVMe drive

LittleBlueBreeze, 2026-10-09:
I'm having an issue with sleep as well, on Ubuntu 26.04.1. Im on the full UEFI firmware, latest version 2609.1.

My issue is that when I suspend the computer, it takes sometimes up to a minute to wake back up.

Here's some output running `journalctl` from a moment I clicked suspend until the computer woke back up (I had the lid open and was pressing spacebar and other keys to try and get it to wake back up the entire time):

```
Oct 09 01:16:01 device-Morphius systemd-logind[2168]: The system will suspend now!
Oct 09 01:16:01 device-Morphius NetworkManager[2228]: <info>  [1791522961.5788] manager: sleep: sleep requested (sleeping: no  enabled: yes)
Oct 09 01:16:01 device-Morphius systemd-logind[2168]: The system will suspend now!
Oct 09 01:16:01 device-Morphius NetworkManager[2228]: <info>  [1791522961.5788] manager: sleep: sleep requested (sleeping: no  enabled: yes)
Oct 09 01:16:01 device-Morphius NetworkManager[2228]: <info>  [1791522961.5799] manager: NetworkManager state is now DISABLED (ASLEEP)
Oct 09 01:16:01 device-Morphius ModemManager[2317]: <msg> [sleep-monitor-systemd] system is about to suspend
Oct 09 01:16:01 device-Morphius NetworkManager[2228]: <info>  [1791522961.5802] device (wlp1s0): state change: activated -> deactivating (reason 'sleeping', m>
Oct 09 01:16:01 device-Morphius ModemManager[2317]: <msg> [sleep-monitor-systemd] ready to sleep; dropping inhibitor
Oct 09 01:16:01 device-Morphius rtkit-daemon[3071]: Demoting known real-time threads.
Oct 09 01:16:01 device-Morphius rtkit-daemon[3071]: Successfully demoted thread 3406 of process 3311.
Oct 09 01:16:01 device-Morphius rtkit-daemon[3071]: Successfully demoted thread 3176 of process 3069.
Oct 09 01:16:01 device-Morphius rtkit-daemon[3071]: Successfully demoted thread 3069 of process 3069.
Oct 09 01:16:01 device-Morphius rtkit-daemon[3071]: Successfully demoted thread 3092 of process 3074.
Oct 09 01:16:01 device-Morphius rtkit-daemon[3071]: Successfully demoted thread 3074 of process 3074.
Oct 09 01:16:01 device-Morphius rtkit-daemon[3071]: Successfully demoted thread 3087 of process 3023.
Oct 09 01:16:01 device-Morphius rtkit-daemon[3071]: Successfully demoted thread 3023 of process 3023.
Oct 09 01:16:01 device-Morphius rtkit-daemon[3071]: Demoted 7 threads.
Oct 09 01:16:01 device-Morphius dbus-daemon[2138]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm>
Oct 09 01:16:01 device-Morphius systemd[1]: Starting NetworkManager-dispatcher.service - Network Manager Script Dispatcher Service...
Oct 09 01:16:01 device-Morphius dbus-daemon[2138]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Oct 09 01:16:01 device-Morphius systemd[1]: Started NetworkManager-dispatcher.service - Network Manager Script Dispatcher Service.
Oct 09 01:16:01 device-Morphius kernel: wlp1s0: deauthenticating from 84:a3:29:98:3b:a6 by local choice (Reason: 3=DEAUTH_LEAVING)
Oct 09 01:16:01 device-Morphius wpa_supplicant[2230]: nl80211: send_event_marker failed: Source based routing not supported
Oct 09 01:16:01 device-Morphius wpa_supplicant[2230]: wlp1s0: CTRL-EVENT-DISCONNECTED bssid=84:a3:29:98:3b:a6 reason=3 locally_generated=1
Oct 09 01:16:01 device-Morphius wpa_supplicant[2230]: wlp1s0: CTRL-EVENT-DSCP-POLICY clear_all
Oct 09 01:16:01 device-Morphius wpa_supplicant[2230]: wlp1s0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Oct 09 01:16:01 device-Morphius NetworkManager[2228]: <info>  [1791522961.7094] device (wlp1s0): state change: deactivating -> disconnected (reason 'sleeping'>
Oct 09 01:16:01 device-Morphius NetworkManager[2228]: <info>  [1791522961.7104] dhcp4 (wlp1s0): canceled DHCP transaction
Oct 09 01:16:01 device-Morphius avahi-daemon[2136]: Withdrawing address record for 2600:4041:5c13:3300:376f:f2f4:a752:312 on wlp1s0.
Oct 09 01:16:01 device-Morphius NetworkManager[2228]: <info>  [1791522961.7105] dhcp4 (wlp1s0): activation: beginning transaction (timeout in 45 seconds)
Oct 09 01:16:01 device-Morphius avahi-daemon[2136]: Withdrawing address record for 2600:4041:5c13:3300:ed5:7cb4:40ff:579a on wlp1s0.
Oct 09 01:16:01 device-Morphius NetworkManager[2228]: <info>  [1791522961.7105] dhcp4 (wlp1s0): state changed no lease
Oct 09 01:16:01 device-Morphius avahi-daemon[2136]: Leaving mDNS multicast group on interface wlp1s0.IPv6 with address 2600:4041:5c13:3300:ed5:7cb4:40ff:579a.
Oct 09 01:16:01 device-Morphius NetworkManager[2228]: <info>  [1791522961.7108] dhcp6 (wlp1s0): canceled DHCP transaction
Oct 09 01:16:01 device-Morphius avahi-daemon[2136]: Joining mDNS multicast group on interface wlp1s0.IPv6 with address fe80::70a3:9aa2:f785:aa84.
Oct 09 01:16:01 device-Morphius NetworkManager[2228]: <info>  [1791522961.7108] dhcp6 (wlp1s0): activation: beginning transaction (timeout in 45 seconds)
Oct 09 01:16:01 device-Morphius avahi-daemon[2136]: Registering new address record for fe80::70a3:9aa2:f785:aa84 on wlp1s0.*.
Oct 09 01:16:01 device-Morphius NetworkManager[2228]: <info>  [1791522961.7108] dhcp6 (wlp1s0): state changed no lease
Oct 09 01:16:01 device-Morphius avahi-daemon[2136]: Withdrawing address record for fe80::70a3:9aa2:f785:aa84 on wlp1s0.
Oct 09 01:16:01 device-Morphius avahi-daemon[2136]: Leaving mDNS multicast group on interface wlp1s0.IPv6 with address fe80::70a3:9aa2:f785:aa84.
Oct 09 01:16:01 device-Morphius avahi-daemon[2136]: Interface wlp1s0.IPv6 no longer relevant for mDNS.
[... cut ...]
Oct 09 01:16:59 device-Morphius wpa_supplicant[2230]: wlp1s0: CTRL-EVENT-REGDOM-BEACON-HINT after freq=5200 max_tx_power=2000
Oct 09 01:17:00 device-Morphius wpa_supplicant[2230]: wlp1s0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
Oct 09 01:17:00 device-Morphius wpa_supplicant[2230]: wlp1s0: CTRL-EVENT-REGDOM-BEACON-HINT before freq=5220 max_tx_power=2000 no_ir=1
Oct 09 01:17:00 device-Morphius wpa_supplicant[2230]: wlp1s0: CTRL-EVENT-REGDOM-BEACON-HINT after freq=5220 max_tx_power=2000
Oct 09 01:17:01 device-Morphius gdm-password][6635]: gkr-pam: unlocked login keyring
```

LittleBlueBreeze, 2026-10-09:
I may have found an acceptable solution: changing the sleep mode from `deep` to `s2idle`. 

Opened a terminal to edit GRUB config file:
```
sudo nano /etc/default/grub
```
add ` mem_sleep_default=s2idle` to the end of this line inside the quotes:
```
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash"
```

I believe `amd_iommu=on` and `iommu=pt` are also recommended for Morphius boards, so the line should look like this before saving:
```
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash amd_iommu=on iommu=pt mem_sleep_default=s2idle"
```

Save the file. Then update GRUB:
```
sudo update-grub
```
Reboot system.

Adding `mem_sleep_default=s2idle` changes the default sleep mode from "deep" to "suspend to idle." I was having issues getting the computer to wake up quickly from `deep` sleep, and `shallow` sleep would just cause the computer not to wake up at all. `s2idle` also caused this once or twice during testing, but pretty rarely. So I think I can live with this.