MrChromebox/firmware #621: Flashing Full ROM/UEFI Thinkpad C13 / Morphius - Device won't boot

ZeroCoolPhD, 2024-04-24:
Hi,

After using the script to flash UEFI/Full rom the device wont boot. Only the power light comes up for a second and does absolutely nothing else. I've managed to unbrick the device following the unbricking instructions using the susy-q cable and flashing the stock firmware backup.  Doing that works and I'm able to get into ChromeOS again.

I've tried flashing the full rom using the script twice to the same result. Also get the same result flashing the full rom with the flashrom tool, following the unbricking instructions.

What could be causing this?

Thanks.

MrChromebox, 2024-04-24:
no idea, boots fine on my Morphius here, and for others. Would need to have you flash a serial-enabled build and log via suzyQ

ZeroCoolPhD, 2024-04-24:
> no idea, boots fine on my Morphius here, and for others. Would need to have you flash a serial-enabled build and log via suzyQ

I know how to flash and open a serial console to see the output, if that is what you are referring to, but what do you mean by serial-enabled build?

MrChromebox, 2024-04-24:
>  what do you mean by serial-enabled build?

a build which has the option to output the boot log in real time via a serial port enabled; one compiled with `CONFILE_CONSOLE_SERIAL=y`

MrChromebox, 2024-04-29:
> I recover the stock rom by CH341 to use chromeos for the time being.

can you try again, and the UEFI image with the ch341a if needed and see if that works?

coolstar tested on BERKNIP and no issues

gladiw, 2024-04-30:
> > I recover the stock rom by CH341 to use chromeos for the time being.
> 
> can you try again, and the UEFI image with the ch341a if needed and see if that works?
> 
> coolstar tested on BERKNIP and no issues

I tried to upgrade 20240415.rom several times  with both utility script and ch341a, and the results were still the same.

fandango96, 2024-05-13:
Hey @ZeroCoolPhD and @gladiw, I'm not sure if this will help but as both of you have the ability to unbrick - may well be worth a try! I have the same device (MORPHIUS) and flashed full ROM / UEFI on it yesterday.

If you've followed the steps here: https://docs.chrultrabook.com/docs/firmware/flashing-firmware.html, once the script says that flash was successful, rebooting will NOT bring up the coreboot screen and the power button will flash like you've observed.

However, if instead, you plug the battery back in (because you must've unplugged it to disable WP) and then hit `Esc + reload + power button` (my charging cable was also connected to the laptop, not sure if this is necessary), you should see the nice coreboot screen and can hit `Esc` to bring up boot options.

ZeroCoolPhD, 2024-05-13:
> However, if instead, you plug the battery back in (because you must've unplugged it to disable WP) and then hit `Esc + reload + power button` (my charging cable was also connected to the laptop, not [...]

@fandango96 This actually worked.

Thanks a lot!

closed 2024-05-13 by ZeroCoolPhD