MrChromebox/firmware #622: Redrix side volume buttons not working on full rom

ChocolateLoverRaj, 2024-04-25:
I'm pretty sure side volume buttons worked on RW_LEGACY. I installed full rom on Redrix, and now the side volume buttons don't work. I'm pretty sure this is a firmware thing and not a software thing because I installed the same exact OS (with the same NixOS configuration) on both Redrix and Jinlon, and Jinlon's side volume buttons work.

crabhit, 2026-09-14:
Reproduced on an HP Elite Dragonfly Chromebook (Redrix). In addition to confirming that EC polling detects the buttons while Linux receives no input events, inspection of the **installed ACPI tables** identifies a missing main-EC interrupt resource that looks relevant to this issue.

### Environment

- Board: Redrix / Brya, Alder Lake PCH-P (`INTC1055`).
- Firmware: **MrChromebox-2603.0**, BIOS date **2026-03-29**.
- OS: AnduinOS 2.0.2 / Ubuntu Resolute; kernel **7.0.0-31-generic**; GNOME desktop.
- EC RO: `redrix_v2.0.26378-d5dba7885d`.
- EC RW: `redrix_v2.0.27860-0f0590ceec`.
- EC build: `redrix_16238.2+tbt5`, `2025-10-12 21:09:39 chrultrabook`.

### Reproduction and observed behavior

Hold and release each physical side volume button while monitoring the dedicated `cros_ec_buttons` and `Intel Virtual Buttons` evdev devices and polling the EC's current button state.

The test used `EC_CMD_MKBP_INFO` (`0x61`, version 1, `EC_MKBP_INFO_CURRENT`, `EC_MKBP_EVENT_BUTTON`) at approximately 10 Hz, with simultaneous non-grabbing evdev reads. It did not change EC settings or consume the MKBP event FIFO.

```text
Elapsed    EC button state
11.316 s   0x2  (volume up pressed)
13.639 s   0x0  (released)
14.346 s   0x4  (volume down pressed)
16.265 s   0x0  (released)

After 60 seconds: 0 KEY_VOLUMEUP / KEY_VOLUMEDOWN events
```

Expected: these presses generate volume-key events and change the desktop volume. Actual: neither a volume change nor an overlay occurs. Both input devices advertise keys 114/115 and report `inhibited = 0`; GNOME's standard volume bindings are present.

The SCI mask is `0x1c6601fb`, already including `EC_HOST_EVENT_MKBP`. This observation alone does not establish whether the EC is generating a corresponding SCI notification.

### Installed ACPI findings and matching source

1. The installed DSDT defines `\_SB.PCI0.LPCB.EC0.CREC` (`GOOG0004`, "EC Command Device") **without `_CRS`**. The installed SSDT also supplies no `_CRS` for this device; there are no dynamically loaded ACPI tables.
2. No main-EC handler is registered at the expected **GSI 103 (`0x67`)**. The existing `chromeos-ec` handler at IRQ 101 belongs to the separate fingerprint controller: the installed SSDT explicitly assigns `0x65` to `\_SB.PCI0.SPI1.CRFP` (`google,cros-ec-spi`).
3. The matching [MrChromebox-2603.0 source](https://github.com/MrChromebox/coreboot/blob/MrChromebox-2603.0/src/ec/google/chromeec/acpi/cros_ec.asl) also omits `_CRS`, carrying [3d45adc: "ec/google/chromeec: drop SYNC IRQ for CREC device"](https://github.com/MrChromebox/coreboot/commit/3d45adc8616a1adc84ca404e23756038a8da7520). This is a carried patch originally authored in 2022, not evidence of a new 2603 regression.
4. The release's [Brya GPIO mapping](https://github.com/MrChromebox/coreboot/blob/MrChromebox-2603.0/src/mainboard/google/brya/variants/baseboard/brya/include/baseboard/gpio.h#L20) defines `EC_SYNC_IRQ = GPP_F17_IRQ`; the applicable [Alder Lake PCH-P definition](https://github.com/MrChromebox/coreboot/blob/MrChromebox-2603.0/src/soc/intel/alderlake/include/soc/gpio_defs.h#L219) gives `0x67`. The board enables the synchronous IRQ and wake support.
5. Current Linux does use this resource: [v7.0 cros_ec_lpc.c](https://github.com/torvalds/linux/blob/v7.0/drivers/platform/chrome/cros_ec_lpc.c#L648) calls `platform_get_irq_optional()`, and [cros_ec.c](https://github.com/torvalds/linux/blob/v7.0/drivers/platform/chrome/cros_ec.c#L246) registers the handler when an IRQ is supplied. Its being optional is consistent with command polling and input-device registration still succeeding.

### Offline validation only; no live fix claimed

A candidate SSDT adding the following resource to the existing `CREC` device compiled successfully:

```asl
Scope (\_SB.PCI0.LPCB.EC0.CREC)
{
    Name (_CRS, ResourceTemplate ()
    {
        Interrupt (ResourceConsumer, Level, ActiveLow, ExclusiveAndWake)
        {
            0x00000067
        }
    })
}
```

Offline ACPICA evaluation with the actual installed DSDT and SSDT returns `AE_NOT_FOUND` for `CREC._CRS`. Adding the candidate creates one additional namespace object without a conflict and returns the expected IRQ resource.

**The candidate has not been installed or boot-tested.** This establishes the missing declaration and offline namespace compatibility, not that restoring it has already fixed event delivery or that suspend/resume works with the candidate. No firmware, boot configuration, or EC settings were changed.

Could the board-appropriate `CREC` interrupt resource be restored or otherwise exposed to Linux, taking the Windows-driver compatibility mentioned in the original removal into account? The local observations provide a concrete missing-resource explanation to investigate alongside the earlier wake-mask hypothesis.

MrChromebox, 2026-11-05:
The missing `CREC` sync IRQ resource should be fixed in **MrChromebox-2609.3** (and later): `CREC` was moved to `\_SB` and the EC sync interrupt is again exposed in `_CRS` for boards that define `EC_ENABLE_SYNC_IRQ` (including Redrix/Brya). That addresses the ACPI gap called out for 2603.x.

Please update to 2609.3 (or newer) and retest the side volume buttons. If they still don’t generate input events, note the installed BIOS version and whether `CREC` now shows an interrupt resource under Linux (`/proc/interrupts` / `cat /sys/bus/acpi/devices/GOOG0004:*/path` and the device’s resources).