MrChromebox/firmware #623: Lenovo Ideapad Flex 3i PUJJO does not load 240417 RW_LEGACY firmware

robcast, 2024-04-27:
I updated the RW_LEGACY firmware for my PUJJO chromebook using the latest firmware-util.sh. It updated the firmware to `rwl_altfw_adl_n-mrchromebox_20240417.bin`.

After a reboot loading the firmware using "select alternate bootloader"/"edk2" failed with an error message and did not load the tianocore loader.

I hacked firmware-util.sh to install the older `rwl_altfw_adl_n-mrchromebox_20230829.bin` which works again.

MrChromebox, 2024-04-27:
hmm that's odd, the only change in the newer version was to add UFS support. I don't have an ADL-N device currently so can't test myself

robcast, 2024-04-28:
When loading the rwl firmware fails I get an unspecific message that there was an error. I can scroll a long "firmware log" in a small window. Does it make sense to try to look for anything specific?

MrChromebox, 2024-04-28:
there's not going to be anything useful there.

there's a bug in ADL boards that required a specific change in the way the RW_LEGACY firmware was built, I'll have to see if that wasn't done here

mariadb-RexJohnston, 2024-05-20:
I can confirm that rwl_altfw_adl_n-mrchromebox_20230829.bin is also currently needed to get edk2 working on a Lenovo 14e Gen 3.  Similar alder lake-n board.

MrChromebox, 2024-05-21:
I've reverted ADL-N to the previous RWL build

closed 2024-05-21 by MrChromebox