MrChromebox/firmware #625: vram issue on yoga c13

haha1121, 2024-04-28:
i follow this guide:https://docs.chrultrabook.com/docs/firmware/compiling-coreboot.html to compile my custom firmware for change the bootsplash image, when i flashed to my laptop, i found out the vram is just 8mb. how should i increase the vram? thank you very much!
[image]

haha1121, 2024-04-28:
and trackpoint cannot work in my custom firmware. any ideas to fix?

MrChromebox, 2024-04-28:
you need to build from a different branch for Picasso boards currently, it will have a working trackpoint, working eMMC and larger UMA:

`git fetch --tags`
`git checkout -b picasso`
`git reset --hard MrChromebox-2402+PCO_eMMC`
`git submodule update --checkout --force`

then apply your changes and rebuild

haha1121, 2024-04-28:
thx man, it's worked!

closed 2024-04-28 by haha1121