MrChromebox/firmware #626: TPM 2.0 with Cr50

ChocolateLoverRaj, 2024-04-28:
It would be really convenient having automatic LUKS unlocking with TPM on Chromebooks, but it doesn't work rn. Are there plans for it to be implemented?

MrChromebox, 2024-04-28:
there's nothing I can do about the fact that the CR50 is not a full TPM 2.0 implementation. I'm not sure if it's sufficient for what you're asking

tlaurion, 2024-11-16:
@MrChromebox https://github.com/tpm2-software/tpm2-tools/issues/3434

MrChromebox, 2024-11-17:
> @MrChromebox [tpm2-software/tpm2-tools#3434](https://github.com/tpm2-software/tpm2-tools/issues/3434)
> 
> Blocker for [linuxboot/heads#1658 (comment)](https://github.com/linuxboot/heads/pull/1658#issuecomment-2136000413) (TPM released Disk Unlock Key: sealing of secret in nvram fails)

@tlaurion CR50 is not a fully TPM 2.0 compliant implementation, as per my comment above. I don't think there's anything missing from the firmware init, other TPM 2.0 chips are fine

tlaurion, 2024-11-20:
> @tlaurion CR50 is not a fully TPM 2.0 compliant implementation, as per my comment above. I don't think there's anything missing from the firmware init, other TPM 2.0 chips are fine

https://github.com/tpm2-software/tpm2-tools/issues/3434#issuecomment-2487391586

Two secrets are sealed with same policy, one succeeds (TPM totp with tpm2), where sealing TPM disk unlock key in seperate nvram region fails.

Two logs provided at

MrChromebox, 2024-11-20:
@tlaurion again I'm not sure what I can do from the firmware init side, or even what you're asking for.

tlaurion, 2024-11-21:
> @tlaurion again I'm not sure what I can do from the firmware init side, or even what you're asking for.

https://github.com/tpm2-software/tpm2-tools/issues/3434#issuecomment-2489935087

Feature missing from tpm2 implementation, so nothing you can do.

jay0lee, 2024-12-02:
Since the Cr50 is not a full implementation, might it be possible to enable Intel PTT on 8th gen and newer Intel CPUs?

https://www.intel.com/content/www/us/en/support/articles/000094205/processors/intel-core-processors.html

unlike a discrete TPM such as the Cr50 or an Infineon chip on a regular PC, my understaning is PTT exists entirely within the Intel CPU/SoC and no additional chip is needed but it must be enabled in firmware.

Is that something Coreboot / Tianacore could enable?

MrChromebox, 2024-12-02:
> might it be possible to enable Intel PTT on 8th gen and newer Intel CPUs?

if the PTT was fused off from the factory (as it is on Chromebooks) via the soft straps then there is no way to re-enable it.

MrChromebox, 2025-12-26:
I'm not sure what you're asking

tlaurion, 2025-12-26:
> > Out of curiosity, is cr50 completely incapable from a hardware standpoint?
> 
> it's is 100% capable. there is even a software flag to enable full TPM 2.0 functionality in the CR50 software. But Google builds the CR50 firmware without it, and the CR50 only runs signed Google firmware, so nothing can be done to modify it

Is there documentation about Google opting out of what is required here to seal unseal a secret in TPM? What that feature set missing would be? There was and is constant interest into adding linuxboot derivative payloads into coreboot with measured boot/verified boot. I will try to revisit logs provided year ago if needed, but I still don't understand why totp can be sealed unsealed, and why tpm luks secondary secret with passphrase as part of policy results in error at sealing attempt in second nvram address.

Is my question clearer this time @MrChromebox?

ChocolateLoverRaj, 2026-07-07:
This did not work for me on ROBO360. `sudo tpm2_createprimary -C o` worked without issues, even without installing ChromeOS. Except when I do 
```
rajas@fedora:~$ sudo systemd-cryptenroll --tpm2-device=auto /dev/mmcblk1p3
🔐 Please enter current passphrase for disk /dev/mmcblk1p3: ••••••••••••••          
WARNING:esys:src/tss2-esys/api/Esys_TestParms.c:272:Esys_TestParms_Finish() Received TPM Error 
ERROR:esys:src/tss2-esys/api/Esys_TestParms.c:96:Esys_TestParms() Esys Finish ErrorCode (0x000b0143) 
TPM device not usable as it does not support the required functionality (AES-128-CFB missing?).
```

codgician, 2026-09-09:
> ## TPM-backed LUKS unlock on Acer Chromebook 315 (BLORB)
> I wanted to document the exact sequence that finally worked for me. This may help others running into the same issue.
> 
> ### Device
> * Acer Chromebook 315 (BLORB)
> * Cr50 / GSC TPM 2.0
> * Latest MrChromebox UEFI Full ROM firmware
> * Fedora Linux 44 (Workstation Edition)
> 
> ### Initial problem
> The TPM was detected correctly:
> 
> tpm2_getcap properties-fixed
> worked without any issues.
> 
> However,
> 
> sudo tpm2_createprimary -C o
> always failed with:
> 
> ```
> TPM_RC_AUTH_FAIL (0x9A2)
> Authorization failure without DA implications
> ```
> 
> The TPM reported:
> 
> ```
> ownerAuthSet:        1
> endorsementAuthSet:  1
> lockoutAuthSet:      1
> ```
> 
> which prevented creating objects in the Owner hierarchy.
> 
> ## What I did
> 1. Restored the Chromebook using the official ChromeOS Recovery image.
> 2. Enabled Developer Mode.
> 3. Opened the developer shell. (Without create an user)
> 4. Set the TPM clear request:
> 
> tpm_clear_request 1
> 5. During troubleshooting I also used `tpm_manager_client`, although I'm not sure whether this was actually required.
> 6. Without performing another ChromeOS setup, I immediately ran the MrChromebox firmware utility and installed the latest UEFI Full ROM firmware.
> 7. Booted Fedora Live.
> 
> At this point the TPM was still in the same state:
> 
> ```
> ownerAuthSet:        1
> endorsementAuthSet:  1
> lockoutAuthSet:      1
> ```
> 
> and
> 
> sudo tpm2_createprimary -C o
> still failed.
> 
> The step that finally solved the issue was:
> 
> sudo tpm2_clear -c p
> Afterwards:
> 
> ```
> ownerAuthSet:        0
> endorsementAuthSet:  0
> lockoutAuthSet:      0
> ```
> 
[... 107 lines cut ...]
    1 : 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    2 : 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    3 : 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    4 : 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    5 : 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    6 : 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    7 : 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    8 : 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    9 : 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    10: 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    11: 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    12: 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    13: 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    14: 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    15: 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    16: 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    17: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    18: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    19: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    20: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    21: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    22: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    23: 0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

```

ChocolateLoverRaj, 2026-09-10:
Basically the TPM2 becomes useless for LUKS unlock because edk2 does not participate in the chain of trust. I wonder if it would be secure if you directly booted Linux as a payload, but I never did that so idk.

tlaurion, 2026-09-10:
> I wonder if it would be secure if you directly booted Linux as a payload, but I never did that so idk.

First Chromebook port (kano) on linuxboot/heads happened last week, having Linux payload participate in the root of trust and taking advantage of tpm2 to seal and unseal tpm sealed disk encryption key with passphrase prior of kexec call and locking chipset to block os for write operations to to firmware. 

Details at https://github.com/linuxboot/heads/pull/2133

ChocolateLoverRaj, 2026-09-10:
@tlaurion  That's really cool. I love how thorough Heads is with security. I might want to put Heads on one of my Chromebooks (Robo360 or Jinlon) some time.

LZeugirdor, 2026-10-08:
@codgician does secure boot not help mitigate this?
@isaiahscape I hope that those parts weren't lasered out. If cr50 won't work I am gonna be stuck typing my password haha
if there's even a partial workaround, i'll take it. I don't feel comfortable needing to unlock my chromebook in public for instance where people can see the password i'm typing

ChocolateLoverRaj, 2026-10-08:
How would secure boot help you unlock your Chromebook without a password?

codgician, 2026-10-08:
> [@codgician](https://github.com/codgician) does secure boot not help mitigate this? [@isaiahscape](https://github.com/isaiahscape) I hope that those parts weren't lasered out. If cr50 won't work I [...]

Secure Boot isn't helpful because PCR7 binding is not working.

tlaurion, 2026-11-01:
For posterity Cr50 and Ti50 have limited subsets of TPM2 and usage needs to comply to be generalized. 

Under heads, this was fixed under https://github.com/linuxboot/heads/pull/2055