MrChromebox/firmware #630: failed: verify it contains a 64-bit UEFI OS

auaulouis, 2024-04-30:
I tried installing mint os on my asus gemini lake 311 chromebook but it gave me "failed: verify it contains a 64-bit UEFI OS" everytime i load the usb. I tried with a sd card too but it doesn't work.

MrChromebox, 2024-04-30:
either you are selecting a device which is not your USB - the error message contains the name of the device it's trying to boot - or you have not properly created a bootable USB. How did you write the Mint ISO file to USB?

MAX2z, 2024-07-31:
I have the same problem i tried mint and ubuntu both shows the same error but the mint when i made it dd it worked but it says error magic number not found and says please launch kernel or something like that and my chrome book storage is 32 is it the storage or i am doing something wrong ? Btw i,m using rufus and i tried 2 usbs .

MrChromebox, 2024-08-01:
the info you provided is too vague to determine what the problem might be. As this isn't a firmware issue -- it's a USB creation one --  your best bet for support is the chrultrabook forums: https://forum.chrultrabook.com

closed 2024-08-01 by MrChromebox