MrChromebox/firmware #633: Samsung ChromeBook XE310XBA

Castaevsky, 2024-05-06:
Hello all.I have Samsung ChromeBook XE310XBA. after booting i see "Crome OS is missing or damaged.

Please remove all connected devices and start recovery" After click TAb i see "HWID: Bluebird C4B-A3D-C3Q-L9E-A8A

recovery_reason:0x5a / 0x5a No bootable storage device in system

VbSD.Flags:0x0003cc40

VbNv.raw: 60 10 00 12 00 02 5a 4c 00 fe ff 00 00 ff ff a5

dev_boot_usb: 0

dev_boot_legacy: 0

dev_default_boot: 0

dev_boot_signed_only: 0

dev_boot_fastboot_full_cap: 0

TMP: fwver=0x00010001 kernver=0x00010001
gbb.recovery_key:73548eec796f801be5d62992ac07efc1155779fc

read-only firmware id: Google_Casta.11297.222.0

active fimware id: Google_Casta.11297.222.0

TPM state:v=1 failed tries=0 max_tries=200"
So, when the recovery process is going on (verifying the integrity of your recovery media...), everything looks okay, but then I see a yellow triangle and an error message saying "An unexpected error has occurred. Please refer to this URL for troubleshooting tips." Does anyone know what to do? Maybe there is a problem with the EMMC, but is it possible to avoid the error described above and install the MCCROMEBOX firmware?
gbb.flags: 0x00000000

gbb.rootkey:5985fd3a58982415348ceae3d8d2c1e127543367

Castaevsky, 2024-05-07:
Unpleasant news. Ok, then I have a few questions. Maybe you can advise me on some step-by-step instructions on how to install the CHROMEBOX firmware? In my possession are two Chromebooks. One of them is an HP G7 11 EE with your custom firmware. I forgot about the EMMC, installing Arch & i3WM on a microSD card and it works perfectly. Maybe you know a solution for how to do something similar on a Samsung Chromebook XE310XBA even with a broken EMCC?

MrChromebox, 2024-05-07:
> recovery_reason:0x5a / 0x5a No bootable storage device in system

is pretty definitive.

>Or this is "your eMMC is dead unfortunately." - death penalty for my chromebook)?

no, like I said, it can be flashed with my UEFI firmware, it might just need a custom build in order to not hang/delay on boot. And will need to be flashed with a SuzyQ or ch341a.

Castaevsky, 2024-05-08:
Спасибо большое) Было приятно прочесть советы от авторитетного человека) Желаю успехов)

closed 2024-05-09 by Castaevsky