MrChromebox/firmware #636: elemi after flash hdd not detected

avdesh-bhandari, 2024-05-10:
hi i tried all ways but after flashing manually or online able to detect HDD in installation please help me out with this issue last  build id 20240222 version, there was no display issue now hardware not detected (storage ) try drivers also bt not work on bios stack its not show 
[image]
[image]

rosanyadav08, 2024-06-05:
Please someone help with this issue

avdesh-bhandari, 2024-06-06:
how to figure out please any one update

MrChromebox, 2025-12-03:
this should be resolved in the 2509.4 release