MrChromebox/firmware #637: Chromebook Flip C302C second USBC (left) port only USB2

jbettcher, 2024-05-13:
Hello - Thanks for all the great work with the firmware script and coreboot configurations for these old chromebooks! I'm getting a great second life out of this little laptop thanks to this project.

A relatively minor but hopefully easy issue - the left USBC port on this laptop running MrChromebox-4.22.4 Release can only "see" USB2 devices. It will work with a USB type C device operating in 2.0 mode (for example, a yubikey 4/5), but any USB 3.0/3.1 device is not recognized at all (for example, a [Cable Matters USB 3.1 flash reader](https://www.amazon.com/gp/product/B01KXWIHZY)). The 3.1 usbc devices do not show at all in any kernel logs when plugged into the left port, despite getting power. I'm running the Full ROM installation with the write protect screw removed, with Ubuntu 23.10. 

The right USBC port does work correctly and can be used for high speed USBC external drives or flash card readers, and provides the ability to boot from a USBC device through the initial bios boot options, so there's a workaround. But it would be great to have the use of the second USBC port for external disks since there are only two ports on this device.

I believe this is just a configuration issue, although this is just a guess. Should the second USBC port here be configured for `USB2_PORT_TYPE_C` or `USB2_PORT_MAX_TYPE_C` ? [overridetree.cb](
https://github.com/MrChromebox/coreboot/blob/9071e69522e38bef841253dc5970ca6b806d2e55/src/mainboard/google/glados/variants/cave/overridetree.cb#L7). 

If there's any way I can test a configuration change or a one-off build I'd love to help. Thanks again for all the work here.

MrChromebox, 2024-05-13:
> I believe this is just a configuration issue, although this is just a guess. Should the second USBC port here be configured for USB2_PORT_TYPE_C or USB2_PORT_MAX_TYPE_C ? [...]

what makes you think it's a configuration issue? The USB section of the board config hasn't changed...ever. And there haven't been any other complaints from other users on the issue. It's more likely a hardware issue with your board I'm afraid. 

Changing the USB_2_ port definition as you suggested won't change anything w/r/t USB_3_ devices not working.

jbettcher, 2024-05-15:
OK interesting. I had read another post of someone having trouble booting USB3 flash drives plugged into the left port on that laptop (which I can’t find right now to link) and had assumed it was some sort of configuration issue with coreboot. It sounds like the left port is working fine for other folks though, so yes I must have a hardware issue. Anyway, thanks for the info - I’ll close this one out

closed 2024-05-15 by jbettcher