MrChromebox/firmware #638: Lenovo ThinkPad C13 Yoga Chromebook (MORPHIUS) fingerprint reader (x-post from the chrultrabook forum)

fandango96, 2024-05-13:
Hey @MrChromebox - thank you for your amazing work. x-post of https://forum.chrultrabook.com/t/lenovo-thinkpad-c13-yoga-chromebook-morphius-fingerprint-reader/2002 as this might be a better place to share things like command outputs.

I flashed the full ROM / UEFI firmware on my C13 Chromebook yesterday and installed EndeavourOS.

Most things seem to be working fine so far - audio worked out of the box, both webcams work, I used https://github.com/saibotd/tp-auto-kbbl as a base for controlling keyboard backlight automatically just like on ChromeOS.

One thing which does not work though is the fingerprint reader - well, I don’t even know if it would work because it’s not even recognized by the system.

```
$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 003 Device 003: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 003 Device 004: ID 04f2:b639 Chicony Electronics Co., Ltd EasyCamera 5M
Bus 003 Device 005: ID 5986:212f Bison Electronics Inc. Integrated Camera
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

I’m wondering if this is an issue with my choice of OS or a firmware issue?

MrChromebox, 2024-05-14:
I'm not aware of any upstream Linux support for the FPR device

fandango96, 2024-05-14:
> I'm not aware of any upstream Linux support for the FPR device

Thanks, would it be possible to expose the device to the OS anyway so it's ready to use when a driver becomes available?

There are a few libfprint forks that support ThinkPad FPRs and although the C13 may be using a different manufacturer, it'd be great to be able to verify that.

MrChromebox, 2024-05-14:
The FPR ACPI device exists and is exposed to the OS

fandango96, 2024-05-14:
Is it this one? snipped from `udevadm info -e`

```
E: PRODUCT=18/27c6/ed3/100
E: NAME="GDIX0000:00 27C6:0ED3 UNKNOWN"
E: PHYS="i2c-GDIX0000:00"
E: UNIQ=""
E: PROP=0
E: EV=9
E: ABS=10000000000
E: MODALIAS=input:b0018v27C6p0ED3e0100-e0,3,kra28,mlsfw
E: USEC_INITIALIZED=4086298
E: ID_INPUT=1
E: ID_PATH=platform-AMDI0010:00
E: ID_PATH_TAG=platform-AMDI0010_00
E: ID_FOR_SEAT=input-platform-AMDI0010_00
E: TAGS=:seat:
E: CURRENT_TAGS=:seat:
```

MrChromebox, 2024-05-14:
> Is it this one? snipped from udevadm info -e

no, the Goodix device is the touchscreen. the FPR is attached to the serial port (UART) and has ACPI device name CRFP

fandango96, 2024-05-14:
Thank you very much, I see it in the list. 

```
P: /devices/platform/AMDI0020:01/AMDI0020:01:0/AMDI0020:01:0.0/serial0/serial0-0
M: serial0-0
R: 0
U: serial
V: cros-ec-uart
E: DEVPATH=/devices/platform/AMDI0020:01/AMDI0020:01:0/AMDI0020:01:0.0/serial0/serial0-0
E: SUBSYSTEM=serial
E: DRIVER=cros-ec-uart
E: MODALIAS=of:NcrfpTCgoogle,cros-ec-uart
```

Also found this open issue for `libfprint` which shows that the feature has been requested already - https://gitlab.freedesktop.org/libfprint/libfprint/-/issues/461

I'll close this issue since it's not a firmware problem, thanks again.

closed 2024-05-14 by fandango96