MrChromebox/firmware #639: BRYA TAEKO halt after calling fsp_raminit

forlhynet, 2024-05-18:
[stocktaeko.zip](https://github.com/MrChromebox/firmware/files/15366267/stocktaeko.zip)
[stock-taeko.txt](https://github.com/MrChromebox/firmware/files/15366293/stock-taeko.txt)
[fulluefi-taeko.txt](https://github.com/MrChromebox/firmware/files/15366295/fulluefi-taeko.txt)
12th ES CPU,The attachments are logs and stock rom
[image]

MrChromebox, 2024-05-19:
I can't tell what repo you've built this from as the version string is very ambiguous: coreboot-24.05-dirty-MrChromebox-4.22.4. You should clear CONFIG_LOCAL_VERSION when building via make.

forlhynet, 2024-05-19:
[log1.txt](https://github.com/MrChromebox/firmware/files/15368314/log1.txt)
I recompiled the firmware on the mainline and captured the logs

forlhynet, 2024-05-19:
The fsp version of stock firmware is v2.2, but the fsp version of new firmware is v2.3. Does this ES CPU not support FSP v2.3?

MrChromebox, 2024-05-19:
I have no idea. ES CPUs aren't supposed to be sold to the public

closed 2025-12-03 by MrChromebox