MrChromebox/firmware #640: Dratini - Volume down button keeps pressing when closing the lid

xperia196, 2024-05-24:
Greetings, thank you for firmware for HP Chromebook c640 G1 (dratini) and most functinos is working fine for me

However, in Linux (Ubuntu, Rocky, etc..), when I lid down my screen, volume down butten keeps pressing.
Sleep works fine, but when I wake again, volume is muted.

Keycode 144 is pressed while lid is closed.

It seems like an ACPI problem, can you help me fix this issue?
Regards,

MrChromebox, 2024-05-25:
that doesn't make a whole lot of sense, how are you determining that the volume button is being pressed?

xperia196, 2024-05-26:
When I close the lid, I hear a volume trigger sound for about 1-2 seconds before the PC goes into sleep.
When I checked with the keyboard event viewer, I confirmed that key code 144 (volume down) was pressed just before the PC went into sleep mode.

xperia196, 2024-05-26:
When I  close the lid, there's 'volume down' sound 1~2 seconds before it goes sleep.
When I checked with the keyboard event viewer, I confirmed that key code 144 (volume down) was pressed just before the PC went into sleep mode.

closed 2026-11-04 by MrChromebox