MrChromebox/firmware #646: Aleena - lost secure boot

fargodwe, 2024-06-10:
Just a quick overview hardware-wise:

- Acer CB315-2H "ALEENA"
- latest firmware update was applied
- usually boot from microSD card (linux)

Recent activity:

- removed microSD card so as not to corrupt
- trying coolstar instructions to attempt to install Windows to internal emmc.  Used rufus.  Booted USB stick and went through installation.  After reboot, got a black screen trying to go to Windows.
- powered off
- inserted microSD card
- powered on, went to boot menu (from jackrabbit screen) and selected the option for my linux

Now getting error about secure boot being disabled.

In boot options screen the secure boot will not set - it stays at standard.

As a result, can't boot the system with any OS.

Any help would be appreciated.

Thanks in advance.

Dave  Eaton

MrChromebox, 2024-06-10:
> Now getting error about secure boot being disabled.

no, you're not. you're getting an error that your device failed to boot, *and* it's reporting the secure boot status

> In boot options screen the secure boot will not set - it stays at standard.
> As a result, can't boot the system with any OS.

what are you expecting it to say other than standard? are you trying to use custom keys?

Also, not sure why you are trying to run Windows on the most underpowered Chromebook possible. A celeron 2955U Chromebook from 2013 runs Windows better.

report the exact error you're getting, not what you think it says.

MrChromebox, 2024-06-10:
again, that msg is just telling you that booting from a particular boot menu option failed, and is reporting the secure boot status along with it. It's not telling you that the boot failed because of the secure boot status. I may have to rework this if it's not clear to others.

'Fedora' is a boot entry that points to a specific file on a specific device. If that device isn't present, can't be read, etc then you get an error. I don't think it's possible to output a more detailed message unfortunately

fargodwe, 2024-06-10:
Yep, got that.  Thing is it was doing it with all media, including media that worked before (microSD card).  Just ended up redoing everything and then booting from a usb stick and reinstalling Fedora to the microSD card and clearing the emmc (chrome, also where I had tried to install Windows prior with the microSD out of the system).  The boot menu didn't even show the SD card or USB stick as an option until I reset everything (via the SuzyQ cable and my other laptop).  Maybe the hardware is finally starting to fail.

On a side note, I do finally have sound in linux.  Had to use Fedora then a special kernel as via weirdtreething on github.

Recovered everything so back to "normal" now.

Thanks

closed 2024-06-10 by MrChromebox