MrChromebox/firmware #648: "verify it contains a 64-bit UEFI OS" shows up 80% of the time

DontEvenTalkToMe, 2024-06-15:
I recently installed GalliumOS on my Acer C720 with a 128 GB expanded drive and the setup went normally. However ever since I have been getting the **"Booting from "galliumos" failed; verify it contains a 64-bit UEFI OS"**. Though if I continuously power the device on and off, eventually it boots and works, however GalliumOS often crashes ~30ish minutes after startup. I am using firmware version: MrChromebox-4.22.4. When I go to the boot menu it shows "galliumos" and then "UEFI Shell" and nothing more, if I am not mistaken it should have my SSD listed?

MrChromebox, 2024-06-16:
GalliumOS hasn't been update or relevant since 2019, please install a current release distro.

the error however sounds like your drive may not be getting detected at boot some times

DontEvenTalkToMe, 2024-06-16:
Thanks for the quick reply! I'll switch my OS and check for if my drive is loose from the inside.

DontEvenTalkToMe, 2024-06-16:
Update: I fixed the issue by opening up my C720 and reconnecting my SSD, hope this helps someone in the future! Closing this issue now.

closed 2024-06-16 by DontEvenTalkToMe