MrChromebox/firmware #649: bugzzy no longer suspends as of 20240615

saurik, 2024-06-19:
I upgraded to the 20240615 release of the firmware on my Samsung Galaxy Chromebook 2 360 (BUGZZY). This resulted in a behavior where it cannot recover from suspend: instead it just turns off and cold boots. I was previously using the 20240223 release, which worked fine. After some messing with the installation scripts, I was able to downgrade to that release, and it corrected the issue. While figuring out how to do that, I also found a firmware release 20240416, which also works fine. I upgraded back to 20240615 to confirm: it doesn't work.

MrChromebox, 2024-06-20:
OS? kernel?

saurik, 2024-06-20:
I am running Arch. I tested kernel versions 6.8.9 and 6.9.5.

MrChromebox, 2024-06-20:
anything interesting in the dmesg log? I'm not able to reproduce here on my JSL test device (maglia)

saurik, 2024-06-21:
I see nothing interesting in journalctl's memory of the dmesg log, but if it is dying as it tries to suspend I can't imagine it would be able to log anything. FWIW, even with the configurations where it does work, all of the dmesg logs aren't saved by journalctl before it goes into suspend, and so the timestamps skew with messages about the suspension process after it comes back.

When I try to wake the computer back up it is just a cold boot.

```
Jun 19 13:50:51 optimist wpa_supplicant[828]: wlp0s20f3: CTRL-EVENT-DSCP-POLICY clear_all
Jun 19 13:50:51 optimist wpa_supplicant[828]: wlp0s20f3: CTRL-EVENT-DSCP-POLICY clear_all
Jun 19 13:50:51 optimist wpa_supplicant[828]: nl80211: deinit ifname=wlp0s20f3 disabled_11b_rates=0
Jun 19 13:50:51 optimist systemd-sleep[5979]: Successfully froze unit 'user.slice'.
Jun 19 13:50:51 optimist systemd-sleep[5979]: Performing sleep operation 'suspend'...
Jun 19 13:50:51 optimist kernel: PM: suspend entry (deep)
-- Boot 1fa31ab4e3e54376be7d49c8f6b3c2f8 --
Jun 19 13:56:47 optimist kernel: Linux version 6.9.5-arch1-1 (linux@archlinux) (gcc (GCC) 14.1.1 20240522, GNU ld (GNU Binutils) 2.42.0) #1 SMP PREEMPT_DYNAMIC Sun, 16 Jun 2024 19:06:37 +0000
```

After spending some time staring at the logs, the only thing I have so far found for you--which I'm hoping might be really useful--is this error message, which is only logged during boots with the new version of coreboot, and even mentions coreboot.

```
Jun 19 11:54:54 optimist kernel: sysfs: cannot create duplicate filename '/bus/platform/devices/simple-framebuffer.0'
Jun 19 11:54:54 optimist kernel: CPU: 0 PID: 430 Comm: (udev-worker) Tainted: G           OE      6.9.5-arch1-1 #1 b9e5462a84a73f67b5c7c6b73f88d2a6349ae768
Jun 19 11:54:54 optimist kernel: Hardware name: Google Bugzzy/Bugzzy, BIOS MrChromebox-4.22.4 04/16/2024
Jun 19 11:54:54 optimist kernel: Call Trace:
Jun 19 11:54:54 optimist kernel:  <TASK>
Jun 19 11:54:54 optimist kernel:  dump_stack_lvl+0x5d/0x80
Jun 19 11:54:54 optimist kernel:  sysfs_warn_dup.cold+0x17/0x23
Jun 19 11:54:54 optimist kernel:  sysfs_do_create_link_sd+0xcf/0xe0
Jun 19 11:54:54 optimist kernel:  bus_add_device+0x6b/0x130
Jun 19 11:54:54 optimist kernel:  device_add+0x3eb/0x8a0
Jun 19 11:54:54 optimist kernel:  platform_device_add+0xed/0x250
Jun 19 11:54:54 optimist kernel:  platform_device_register_full+0xbb/0x140
Jun 19 11:54:54 optimist kernel:  platform_device_register_resndata.constprop.0+0x54/0x80 [framebuffer_coreboot 69e23c792c91d7d333bff41324bb21829862cd65]
Jun 19 11:54:54 optimist kernel:  framebuffer_probe+0x154/0x1a0 [framebuffer_coreboot 69e23c792c91d7d333bff41324bb21829862cd65]
Jun 19 11:54:54 optimist kernel:  really_probe+0xdb/0x340
Jun 19 11:54:54 optimist kernel:  ? pm_runtime_barrier+0x54/0x90
Jun 19 11:54:54 optimist kernel:  ? __pfx___driver_attach+0x10/0x10
Jun 19 11:54:54 optimist kernel:  __driver_probe_device+0x78/0x110
Jun 19 11:54:54 optimist kernel:  driver_probe_device+0x1f/0xa0
Jun 19 11:54:54 optimist kernel:  __driver_attach+0xba/0x1c0
Jun 19 11:54:54 optimist kernel:  bus_for_each_dev+0x8c/0xe0
Jun 19 11:54:54 optimist kernel:  bus_add_driver+0x112/0x1f0
Jun 19 11:54:54 optimist kernel:  driver_register+0x72/0xd0
Jun 19 11:54:54 optimist kernel:  ? __pfx_framebuffer_driver_init+0x10/0x10 [framebuffer_coreboot 69e23c792c91d7d333bff41324bb21829862cd65]
Jun 19 11:54:54 optimist kernel:  do_one_initcall+0x58/0x310
Jun 19 11:54:54 optimist kernel:  do_init_module+0x60/0x220
Jun 19 11:54:54 optimist kernel:  init_module_from_file+0x89/0xe0
Jun 19 11:54:54 optimist kernel:  idempotent_init_module+0x121/0x2b0
Jun 19 11:54:54 optimist kernel:  __x64_sys_finit_module+0x5e/0xb0
Jun 19 11:54:54 optimist kernel:  do_syscall_64+0x83/0x190
Jun 19 11:54:54 optimist kernel:  ? switch_fpu_return+0x4e/0xd0
Jun 19 11:54:54 optimist kernel:  ? syscall_exit_to_user_mode+0x75/0x210
Jun 19 11:54:54 optimist kernel:  ? do_syscall_64+0x8f/0x190
Jun 19 11:54:54 optimist kernel:  ? syscall_exit_to_user_mode+0x75/0x210
Jun 19 11:54:54 optimist kernel:  ? do_syscall_64+0x8f/0x190
Jun 19 11:54:54 optimist kernel:  ? do_user_addr_fault+0x34e/0x620
Jun 19 11:54:54 optimist kernel:  ? clear_bhb_loop+0x45/0xa0
Jun 19 11:54:54 optimist kernel:  ? clear_bhb_loop+0x45/0xa0
Jun 19 11:54:54 optimist kernel:  ? clear_bhb_loop+0x45/0xa0
Jun 19 11:54:54 optimist kernel:  entry_SYSCALL_64_after_hwframe+0x76/0x7e
Jun 19 11:54:54 optimist kernel: RIP: 0033:0x796f25927e9d
Jun 19 11:54:54 optimist kernel: Code: ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 63 de 0c 00 f7 d8 64 89 01 48
Jun 19 11:54:54 optimist kernel: RSP: 002b:00007ffe77e24ab8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
Jun 19 11:54:54 optimist kernel: RAX: ffffffffffffffda RBX: 0000601531e78500 RCX: 0000796f25927e9d
Jun 19 11:54:54 optimist kernel: RDX: 0000000000000004 RSI: 0000796f25e60376 RDI: 0000000000000016
Jun 19 11:54:54 optimist kernel: RBP: 0000796f25e60376 R08: 0000000000000001 R09: 00007ffe77e24b00
Jun 19 11:54:54 optimist kernel: R10: 0000000000000050 R11: 0000000000000246 R12: 0000000000020000
Jun 19 11:54:54 optimist kernel: R13: 0000601531ee7920 R14: 0000000000000000 R15: 0000601531ee9ad0
Jun 19 11:54:54 optimist kernel:  </TASK>
Jun 19 11:54:54 optimist kernel: coreboot: could not register framebuffer
Jun 19 11:54:54 optimist kernel: framebuffer coreboot8: probe with driver framebuffer failed with error -17
```

saurik, 2024-06-21:
Ok, no: that error message wasn't from the boots with the new version of coreboot, they were from the boots with the new version of the kernel, and it is unrelated to the issue (as it happened during even my current boot). I've realized I can look at prior dmesg logs and figure out whether it is a new or old version of coreboot based on different memory regions logged in the BIOS memory map. I can try and look if there are any differences between boots with the same kernel and different versions of coreboot.

So, yeah: the only thing I have for you is "it doesn't work" ;P.

ellyq, 2024-07-05:
I'm afraid you would need to flash debug build and see what happens on resume in this situation (using SuzyQ).
AFAIK neither of us own any JasperLake devices, so support for those is provided based on community feedback.

MrChromebox, 2024-07-05:
> I'm afraid you would need to flash debug build and see what happens on resume in this situation (using SuzyQ). AFAIK neither of us own any JasperLake devices, so support for those is provided based on community feedback.

I actually have a MAGLIA now, and no issues with suspend/resume on Windows or Linux under 2405.0

saurik, 2024-07-07:
@ellyq FWIW, I'll be able to do that at some point in the near future (and want to do that anyway as it is driving me crazy that coreboot refuses to boot if my f2fs SD card is inserted while it initializes) but I'm unsure what I need as far as a "debug build"... are you suggesting I'd want GDB_DEBUG=1?

closed 2026-11-04 by MrChromebox