MrChromebox/firmware #651: hp c1030 i7-10610u/16G，the screen don‘t adjust brightness in win11 after updated the new version"MrChromebox-2405 Release"

wg-cn, 2024-06-26:
hp c1030 i7-10610u/16G，the screen don‘t adjust brightness in win11 after updated the new version"MrChromebox-2405 Release",

adjustment brightness is ok in linux.

MrChromebox, 2024-06-26:
this is a Jinlon device? if so, I have a fix, just away on travel so won't be able to update most likely for another week or so. Fix is in the MrChromebox-2405 branch if you want to build yourself

wg-cn, 2024-06-26:
> this is a Jinlon device? if so, I have a fix, just away on travel so won't be able to update most likely for another week or so. Fix is in the MrChromebox-2405 branch if you want to build yourself

yes, device Board Name is Jinlon.Thanks.
--

MrChromebox, 2024-06-26:
just pushed a new version dated 6/25, see if that fixes the issue for you

wg-cn, 2024-06-28:
> just pushed a new version dated 6/25, see if that fixes the issue for you

thanks,i will try bulid rom myself.

wg-cn, 2024-06-28:
> just pushed a new version dated 6/25, see if that fixes the issue for you

The function is working properly after updated the new ROM, thanks

closed 2024-06-28 by MrChromebox