MrChromebox/firmware #652: USB 3.0 add-on cards have issues with UAS/UASP under Samsung Chromebox Stumpy

ShapeShifter499, 2024-07-01:
I'm running a Samsung Chromebox Stumpy and I tried adding a USB 3.0 card to the unused Mini PCIe slot but anything connected with UAS/UASP (USB Attached SCSI Protocol) tends to throw errors.  I am at my wits end trying to figure out where the issue is. Enclosures that have UASP work fine when connected to a Surface Pro 5 I have running Linux. 

I'm wondering if possibly the Coreboot build is missing some UASP specific settings.

So far I've tried
UPD720202 Renesas based Mini PCIe  card connected to a Mini PCIe flat ribbon extender.
Frisco Logic FL1100 based PCIe card connected to a PCIe 1x to Mini PCIe adapter.

In both cases I see these sort of messages. Enclosure here is a Nexstar HX.
```
[  243.863681] sd 7:0:0:0: [sdc] tag#26 uas_eh_abort_handler 0 uas-tag 23 inflight: CMD IN 
[  243.863690] sd 7:0:0:0: [sdc] tag#26 CDB: Read(16) 88 00 00 00 00 01 4e c0 98 d0 00 00 01 58 00 00
[  243.863852] sd 7:0:0:0: [sdc] tag#25 uas_eh_abort_handler 0 uas-tag 22 inflight: CMD IN 
[  243.863854] sd 7:0:0:0: [sdc] tag#25 CDB: Read(16) 88 00 00 00 00 01 4e c0 98 30 00 00 00 70 00 00
[  243.863986] sd 7:0:0:0: [sdc] tag#24 uas_eh_abort_handler 0 uas-tag 21 inflight: CMD IN 
[  243.863988] sd 7:0:0:0: [sdc] tag#24 CDB: Read(16) 88 00 00 00 00 01 4e c0 98 08 00 00 00 08 00 00
[  243.864115] sd 7:0:0:0: [sdc] tag#23 uas_eh_abort_handler 0 uas-tag 20 inflight: CMD IN 
[  243.864117] sd 7:0:0:0: [sdc] tag#23 CDB: Read(16) 88 00 00 00 00 01 4e c0 97 e8 00 00 00 08 00 00
[  243.864238] sd 7:0:0:0: [sdc] tag#22 uas_eh_abort_handler 0 uas-tag 19 inflight: CMD IN 
[  243.864240] sd 7:0:0:0: [sdc] tag#22 CDB: Read(16) 88 00 00 00 00 01 4e c0 97 90 00 00 00 50 00 00
[  243.864343] sd 7:0:0:0: [sdc] tag#21 uas_eh_abort_handler 0 uas-tag 18 inflight: CMD IN 
[  243.864345] sd 7:0:0:0: [sdc] tag#21 CDB: Read(16) 88 00 00 00 00 01 4e c0 97 78 00 00 00 08 00 00
[  243.864472] sd 7:0:0:0: [sdc] tag#20 uas_eh_abort_handler 0 uas-tag 17 inflight: CMD IN 
[  243.864474] sd 7:0:0:0: [sdc] tag#20 CDB: Read(16) 88 00 00 00 00 01 4e c0 97 38 00 00 00 38 00 00
[  243.864613] sd 7:0:0:0: [sdc] tag#19 uas_eh_abort_handler 0 uas-tag 16 inflight: CMD IN 
[  243.864626] sd 7:0:0:0: [sdc] tag#19 CDB: Read(16) 88 00 00 00 00 01 4e c0 90 00 00 00 03 50 00 00
[  243.864763] sd 7:0:0:0: [sdc] tag#18 uas_eh_abort_handler 0 uas-tag 15 inflight: CMD IN 
[  243.864770] sd 7:0:0:0: [sdc] tag#18 CDB: Read(16) 88 00 00 00 00 01 4e c0 8f c0 00 00 00 38 00 00
[  243.864914] sd 7:0:0:0: [sdc] tag#17 uas_eh_abort_handler 0 uas-tag 14 inflight: CMD IN 
[  243.864921] sd 7:0:0:0: [sdc] tag#17 CDB: Read(16) 88 00 00 00 00 01 4e c0 8d 18 00 00 02 90 00 00
[  243.865066] sd 7:0:0:0: [sdc] tag#16 uas_eh_abort_handler 0 uas-tag 13 inflight: CMD IN 
[  243.865067] sd 7:0:0:0: [sdc] tag#16 CDB: Read(16) 88 00 00 00 00 01 4e c0 8d 00 00 00 00 08 00 00
[  243.865196] sd 7:0:0:0: [sdc] tag#15 uas_eh_abort_handler 0 uas-tag 10 inflight: CMD IN 
[  243.865202] sd 7:0:0:0: [sdc] tag#15 CDB: Read(16) 88 00 00 00 00 01 4e c0 89 b8 00 00 00 08 00 00
[  243.865297] sd 7:0:0:0: [sdc] tag#14 uas_eh_abort_handler 0 uas-tag 9 inflight: CMD 
[  243.865308] sd 7:0:0:0: [sdc] tag#14 CDB: Read(16) 88 00 00 00 00 01 4e c0 89 10 00 00 00 a0 00 00
[  243.865310] sd 7:0:0:0: [sdc] tag#13 uas_eh_abort_handler 0 uas-tag 12 inflight: CMD IN 
[  243.865311] sd 7:0:0:0: [sdc] tag#13 CDB: Read(16) 88 00 00 00 00 01 4e c0 8c f0 00 00 00 08 00 00
[  243.865439] sd 7:0:0:0: [sdc] tag#12 uas_eh_abort_handler 0 uas-tag 11 inflight: CMD IN 
[  243.865445] sd 7:0:0:0: [sdc] tag#12 CDB: Read(16) 88 00 00 00 00 01 4e c0 89 d0 00 00 02 80 00 00
[  243.865585] sd 7:0:0:0: [sdc] tag#10 uas_eh_abort_handler 0 uas-tag 8 inflight: CMD 
[  243.865592] sd 7:0:0:0: [sdc] tag#10 CDB: Read(16) 88 00 00 00 00 01 4e 80 a2 80 00 00 00 08 00 00
[  243.865599] sd 7:0:0:0: [sdc] tag#9 uas_eh_abort_handler 0 uas-tag 7 inflight: CMD 
[  243.865605] sd 7:0:0:0: [sdc] tag#9 CDB: Read(16) 88 00 00 00 00 01 4e 80 a2 50 00 00 00 08 00 00
[  243.865620] sd 7:0:0:0: [sdc] tag#8 uas_eh_abort_handler 0 uas-tag 6 inflight: CMD 
[  243.865622] sd 7:0:0:0: [sdc] tag#8 CDB: Read(16) 88 00 00 00 00 01 4e 80 a2 28 00 00 00 08 00 00
[  243.865624] sd 7:0:0:0: [sdc] tag#7 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD IN 
[  243.865625] sd 7:0:0:0: [sdc] tag#7 CDB: Read(16) 88 00 00 00 00 01 4e 80 9f 50 00 00 00 70 00 00
[  243.880369] scsi host7: uas_eh_device_reset_handler start
[  244.003992] usb 4-2: reset SuperSpeed USB device number 3 using xhci_hcd
[  244.022473] scsi host7: uas_eh_device_reset_handler success
```

Stumpy running MrChromebox-4.22.4 Coreboot build.  
Arch Linux
16GB RAM
512GB SSD

MrChromebox, 2024-07-01:
cbmem log please

ShapeShifter499, 2024-07-03:
Coreboot updated to MrChromebox-2405.0

cbmem log with the FL1100 card inserted.

```
[kumo-chromebox ~]# cbmem -1
*** Pre-CBMEM romstage console overflowed, log truncated! ***
tatic northbridge registers... done
[DEBUG]  Initializing Graphics...
[DEBUG]  Back from systemagent_early_init()
[INFO ]  Intel ME early init
[INFO ]  Intel ME firmware is ready
[DEBUG]  ME: Requested 8MB UMA
[DEBUG]  Starting native Platform init
[DEBUG]  DMI: Running at X4 @ 5000MT/s
[DEBUG]  FMAP: area RW_MRC_CACHE found @ 500000 (65536 bytes)
[DEBUG]  Trying stored timings.
[DEBUG]  Starting Sandy Bridge RAM training (fast boot).
[DEBUG]  100MHz reference clock support: no
[DEBUG]  PLL_REF100_CFG value: 0x0
[DEBUG]  Trying CAS 9, tCK 384.
[DEBUG]  Found compatible clock, CAS pair.
[DEBUG]  Selected DRAM frequency: 666 MHz
[DEBUG]  Selected CAS latency   : 9T
[DEBUG]  MPLL busy... done in 40 us
[DEBUG]  MPLL frequency is set at : 666 MHz
[DEBUG]  Done dimm mapping
[DEBUG]  Update PCI-E configuration space:
[DEBUG]  PCI(0, 0, 0)[a0] = 0
[DEBUG]  PCI(0, 0, 0)[a4] = 2
[DEBUG]  PCI(0, 0, 0)[bc] = 86a00000
[DEBUG]  PCI(0, 0, 0)[a8] = 78e00000
[DEBUG]  PCI(0, 0, 0)[ac] = 2
[DEBUG]  PCI(0, 0, 0)[b8] = 80000000
[DEBUG]  PCI(0, 0, 0)[b0] = 80a00000
[DEBUG]  PCI(0, 0, 0)[b4] = 80800000
[DEBUG]  PCI(0, 0, 0)[7c] = 7f
[DEBUG]  PCI(0, 0, 0)[70] = ff800000
[DEBUG]  PCI(0, 0, 0)[74] = 1
[DEBUG]  PCI(0, 0, 0)[78] = ff800c00
[DEBUG]  Done memory map
[DEBUG]  Done io registers
[DEBUG]  t123: 1768, 9120, 500
[NOTE ]  ME: FWS2: 0x164e0006
[NOTE ]  ME:  Bist in progress: 0x0
[NOTE ]  ME:  ICC Status      : 0x3
[NOTE ]  ME:  Invoke MEBx     : 0x0
[NOTE ]  ME:  CPU replaced    : 0x0
[NOTE ]  ME:  MBP ready       : 0x0
[NOTE ]  ME:  MFS failure     : 0x0
[NOTE ]  ME:  Warm reset req  : 0x0
[NOTE ]  ME:  CPU repl valid  : 0x0
[NOTE ]  ME:  (Reserved)      : 0x0
[NOTE ]  ME:  FW update req   : 0x0
[NOTE ]  ME:  (Reserved)      : 0x0
[NOTE ]  ME:  Current state   : 0x4e
[NOTE ]  ME:  Current PM event: 0x6
[NOTE ]  ME:  Progress code   : 0x1
[NOTE ]  Waited long enough, or CPU was not replaced, continue...
[NOTE ]  PASSED! Tell ME that DRAM is ready
[NOTE ]  ME: FWS2: 0x16500006
[NOTE ]  ME:  Bist in progress: 0x0
[NOTE ]  ME:  ICC Status      : 0x3
[NOTE ]  ME:  Invoke MEBx     : 0x0
[NOTE ]  ME:  CPU replaced    : 0x0
[NOTE ]  ME:  MBP ready       : 0x0
[NOTE ]  ME:  MFS failure     : 0x0
[NOTE ]  ME:  Warm reset req  : 0x0
[NOTE ]  ME:  CPU repl valid  : 0x0
[NOTE ]  ME:  (Reserved)      : 0x0
[... 784 lines cut ...]
[DEBUG]  TCPA TCGLOG12. 0x7fe20000 0x00010000
[DEBUG]  SMBIOS     13. 0x7fe18000 0x00008000
[DEBUG]  IMD small region:
[DEBUG]    IMD ROOT    0. 0x7fffec00 0x00000400
[DEBUG]    RO MCACHE   1. 0x7fffe920 0x000002e0
[DEBUG]    FMAP        2. 0x7fffe7e0 0x00000134
[DEBUG]    ROMSTAGE    3. 0x7fffe7c0 0x00000004
[DEBUG]    ROMSTG STCK 4. 0x7fffe700 0x000000a8
[DEBUG]    ACPI GNVS   5. 0x7fffe600 0x00000100
[DEBUG]    TPM PPI     6. 0x7fffe4a0 0x0000015a
[DEBUG]  BS: BS_WRITE_TABLES run times (exec / console): 2 / 0 ms
[INFO ]  CBFS: Found 'fallback/payload' @0x45b80 size 0x13ed26 in mcache @0x7fffeb90
[DEBUG]  Checking segment from ROM address 0xffd99dac
[DEBUG]  Checking segment from ROM address 0xffd99dc8
[DEBUG]  Loading segment from ROM address 0xffd99dac
[DEBUG]    code (compression=1)
[DEBUG]    New segment dstaddr 0x00800000 memsize 0x1000000 srcaddr 0xffd99de4 filesize 0x13ecee
[DEBUG]  Loading Segment: addr: 0x00800000 memsz: 0x0000000001000000 filesz: 0x000000000013ecee
[DEBUG]  using LZMA
[DEBUG]  Loading segment from ROM address 0xffd99dc8
[DEBUG]    Entry Point 0x008016b3
[DEBUG]  BS: BS_PAYLOAD_LOAD run times (exec / console): 436 / 0 ms
[DEBUG]  ICH-NM10-PCH: watchdog disabled
[DEBUG]  Jumping to boot code at 0x008016b3(0x7fe54000)
```

MrChromebox, 2024-07-03:
since you have both working/non-working setups, can you dump the info from `sudo lspci -vvv -s <pci bus:device.function>` for both cases? attach as files vs pasting if possible

ShapeShifter499, 2024-07-03:
@MrChromebox dump the 'working' setup from the surface pro?  That doesn't use any pci cards, the USB 3.0 is built in there.   On the Chromebox, USB devices error if they use UASP.

MrChromebox, 2024-07-03:
ok, just dump it from the Chromebox then, will see if anything jumps out

ShapeShifter499, 2024-07-03:
Here are my logs, including a new third PCI card I brought for testing.  But now I'm even more confused as to what is screwing up here. The Renesas based card stopped throwing errors for the UASP enabled hard drive enclosure I am using. I even tested with a high I/O load by mounting the drive and running various versions of `dd if=/dev/urandom of=/mnt/test_urandom.img` for a bit, I even tried three writes to different test files at the same time with `dd` and still no errors.

The UASP errors like the ones seen with the other two cards was the whole reason I went and tried any other cards than the Renesas card. Do you have any ideas what might have happened?  The only variables here I can think of. I updated both the kernel and the coreboot build. And that I was swapping cards around, so possibly dodgy connection? Unfortunately I didn't check any lspci outputs when the Renesas card was throwing errors.

I do know, no matter how many times I tried to seat the Fresco Logic and ASMedia cards that I could reliably get those UASP errors.

**uPD720202 Renesas based Mini PCIe card connected to a Mini PCIe flat ribbon extender. Included two sets of logs, 'boot2' is after a reboot command.**
[Renesas_uPD720202-cbmem_log.txt](https://github.com/user-attachments/files/16087147/Renesas_uPD720202-cbmem_log.txt)
[Renesas_uPD720202-dmesg_log.txt](https://github.com/user-attachments/files/16087152/Renesas_uPD720202-dmesg_log.txt)
[Renesas_uPD720202-lspci.txt](https://github.com/user-attachments/files/16087155/Renesas_uPD720202-lspci.txt)
[Renesas_uPD720202-lspci_extended.txt](https://github.com/user-attachments/files/16087157/Renesas_uPD720202-lspci_extended.txt)

[Renesas_uPD720202-boot2-cbmem_log.txt](https://github.com/user-attachments/files/16087159/Renesas_uPD720202-boot2-cbmem_log.txt)
[Renesas_uPD720202-boot2-dmesg_log.txt](https://github.com/user-attachments/files/16087160/Renesas_uPD720202-boot2-dmesg_log.txt)
[Renesas_uPD720202-boot2-lspci.txt](https://github.com/user-attachments/files/16087162/Renesas_uPD720202-boot2-lspci.txt)
[Renesas_uPD720202-boot2-lspci_extended.txt](https://github.com/user-attachments/files/16087165/Renesas_uPD720202-boot2-lspci_extended.txt)

**Fresco Logic FL1100 based PCIe card connected to a PCIe 1x to Mini PCIe adapter, which is then connected to the Mini PCIe flat ribbon extender.**
[Fresco_Logic_FL1100-cbmem_log.txt](https://github.com/user-attachments/files/16085556/Fresco_Logic_FL1100-cbmem_log.txt)
[Fresco_Logic_FL1100-dmesg_log.txt](https://github.com/user-attachments/files/16085559/Fresco_Logic_FL1100-dmesg_log.txt)
[Fresco_Logic_FL1100-lspci.txt](https://github.com/user-attachments/files/16085564/Fresco_Logic_FL1100-lspci.txt)
[Fresco_Logic_FL1100-lspci_extended.txt](https://github.com/user-attachments/files/16085568/Fresco_Logic_FL1100-lspci_extended.txt)

**ASMedia ASM1042 based PCIe card connected to a PCIe 1x to Mini PCIe adapter, which is then connected to the Mini PCIe flat ribbon extender.**
[ASMedia_ASM1042-cbmem_log.txt](https://github.com/user-attachments/files/16085543/ASMedia_ASM1042-cbmem_log.txt)
[ASMedia_ASM1042-dmesg_log.txt](https://github.com/user-attachments/files/16085545/ASMedia_ASM1042-dmesg_log.txt)
[ASMedia_ASM1042-lspci.txt](https://github.com/user-attachments/files/16085551/ASMedia_ASM1042-lspci.txt)
[ASMedia_ASM1042-lspci_extended.txt](https://github.com/user-attachments/files/16085555/ASMedia_ASM1042-lspci_extended.txt)

ShapeShifter499, 2024-07-08:
@MrChromebox I'm not sure if the Coreboot update happened to carry a fix, or if it was the kernel, or if simply reseating all of the connections on the board again helped. 

But it's been a few days of 24/7 uptime and no UAS errors so far. I've gone back to using the uPD720202 Renesas based Mini PCIe card.

Do you have any thoughts on the matter before this gets closed?

MrChromebox, 2024-07-08:
not really, if the issue comes back we'll likely need to dig into the PCI transactions to see what's happening

ShapeShifter499, 2024-08-29:
I'm going to close this. I no longer experience issues and any new UAS/UASP issues I have now appear to be related to two failing hard drives I have. The still working hard drives I have no longer crash the system when used with an enclosure that appears to properly implement UAS/UASP. 

However the uPD720202 Renesas based Mini PCIe card seems to be the only card I brought that works with this system.

Anyone can reopen this if they experience issues related on this model Samsung Chromebox Stumpy

closed 2024-08-29 by ShapeShifter499