MrChromebox/firmware #653: HP 14 G5 Chromebooks - TianoCore "beeps" but doesn't "engage"

clm1919, 2024-07-05:
First thank you for all the work you've done and this script, it's encouraged me to try Linux again after many (MANY) years.

after several failed attempts (and different Debian and Arch usb ISO's), I finally got an intel based HP 14 G5 (ApolloLake) to work using a minimal Live Debian USB w/ xfce to boot and (mostly) work. 

**However the HP14A G5 (AMD cpu) won't.  I get to the CTRL-L boot screen, but when I hit "2" (TianoCore) it just BEEPS.**

When I try option one (on either device) I always got some message about USB addresses not being where expected and all the current online advice said "Use TianoCore".

I did note that when I ran your utility script on the INTEL machine I got a second Y/N option about setting usb boot priorities that I don't get when I try it on the AMD based machine.

I read through some of the other "issues" and you commented that some AMD based machines just won't work (boo GOOGLE).

Should I stop banging my head on this HP15A?  Is it probably just a dead end?

I have a Lenovo Yoga 11e (G3 - Ultima board) I could try, but it's the one chromebook I actually USE, despite it being no longer updated.  I'd hate to break it.

MrChromebox, 2024-07-05:
CAREENA? Tianocore/edk2 is known to be broken due to issues with the stock Google firmware. I might be able to get SeaBIOS working, but it would take some effort that I'm not sure is worth it given the poor state of OS support AMD Stoneyridge devices have

clm1919, 2024-07-05:
I figured as much - the AMD chromebook at least has sound (under ChromeOS), I so can use it to play media files.

Thanks for the fast reply - now I can stop driving myself crazy thinking that i'm doing something (else) wrong :-)

Can't get sound working on the Intel 14G5, but everything else works - it's installed on the internal now (no more ChromeOS) - couldn't get it to install to the SD card - but maybe i can put a swap partition on the SD card.  Makes a nice fast 2nd monitor/web-browser when I need to conserve resources on my other machines.

THANKS again for all your hard work and organization of information!

MrChromebox, 2024-07-05:
You flashed the UEFI Full ROM firmware on it? You should be able to get audio working under Linux, the chrultrabook project has some more info on that -- needs a custom kernel: https://docs.chrultrabook.com/docs/firmware/supported-devices.html

MrChromebox, 2024-07-05:
closing since this is a can't-fix

closed 2024-07-05 by MrChromebox