MrChromebox/firmware #654: The ubuntu bootup drops into a initramfs prompt

cparman5, 2024-07-06:
I have a HP Chromebook 14 G5 with a 15G internal SSD drive. It is too small to install Ubuntu along with ChromeOS so I installed Ubuntu on a 512G Micro SD card. Now my issue is when choose the MicroSD from the BusyBox list the Ubuntu bootup drops into a initramfs prompt. Is there a way to tell the Chromebook to bootup the Ubuntu installation I installed on the MicroSD card? Thank you in advance for your help. 

By the way, the name of the microSD card relative to what chromeOS sees it as is called mmcblk0, that's where I told the Ubuntu installer to install Ubuntu.

MrChromebox, 2024-07-06:
so to clarify, you're using the RW_LEGACY firmware on what device? SNAPPY?

Does the SD card show up as a boot option in the legacy boot menu?

mmcblk0 is normally the internal eMMC.

cparman5, 2024-07-06:
The device is SNAPPY. The SD card shows up as as a boot option in the legacy menu. The normal internal eMMC is labeled mmcblk1.

MrChromebox, 2024-07-06:
this doesn't sound like a firmware issue then

cparman5, 2024-07-06:
I experimented with installing gallium Linux on the chromebooks main internal (labeled mmcblk1) and gallium Linux boots fine. My overall question is does your firmware-util allow for booting Linux on a secondary drive (in my case a micoSD card)? If not, I am cool with that. I just need to clear on the limitations of the firmware-util script tool :)

MrChromebox, 2024-07-06:
this isn't a limitation of the RW_LEGACY firmware for your device, if the device shows up in the boot menu, then the firmware has done it's job. If you want, can test a Linux ISO on another SD card and see if that boots

cparman5, 2024-07-06:
Thank you for the reply. I have another query, in my experiments I have always chosen the Option #1(Install/Update RW_LEGACY Firmware) and using Chrx to install Ubuntu Linux. Would you surmise that using the option in Chrx called (-t TARGET) thus using the device (mmcblk0) as the target (E.g., -t /dev/mmcblk0). based on my comprehension of the Chrx tool this string "chrx -d Ubuntu -e standard -r lts -t /dev/mmcblk0" could possibly boot from the external SD card. What do you think?

MrChromebox, 2024-07-06:
chrx hasn't been updated in 5+ years, it is irrelevant and should not be used.

Boot a current 2024 distro from USB, then install to the SD card. Most likely chrx simply installed a very old version of an OS that doesn't support your hardware.

cparman5, 2024-07-06:
Can you clarify this statement per the firmware-util.sh script. 

####################

Default from boot from USB?

If N, always boot from internal storage unless selected from boot menu. [y/N]

#####################

This question confuses me. 

For the way I want to install Linux (on an external SD card and which is what you stated in your previous reply), what would you choose? Yes or No

cparman5, 2024-07-07:
Hello Sir:

I have an update. I performed another dual boot ChromeOS and Ubuntu install experiment. Here are the results.

1. After the install of Ubuntu on the MicroSD card (while is was inserted into the MicroSD Slot and then restarted the Chromebook, the Ubuntu installation would not bootup.
2. On a hunch, I have a SD adapter *(that accepts different form factors of SD cards, even the MicroSD factor size). So I inserted the MicroSD card into it and plugged the SD card adapter into a USB port. This time Ubuntu booted up fine.

Conclusion: I believe there might be a limitation with respect to your firmware-util.sh script that doesn't allow a Linux install on a MicroSD card to boot when the MicroSD card is inserted into the MicroSD slot of the Chromebook.

MrChromebox, 2024-07-07:
> Hello Sir:
> 
> I have an update. I performed another dual boot ChromeOS and Ubuntu install experiment. Here are the results.
> 
> 1. After the install of Ubuntu on the MicroSD card (while is was inserted into the MicroSD Slot and then restarted the Chromebook, the Ubuntu installation would not bootup.

wouldn't boot **how**?

> 2. On a hunch, I have a SD adapter *(that accepts different form factors of SD cards, even the MicroSD factor size). So I inserted the MicroSD card into it and plugged the SD card adapter into a USB port. This time Ubuntu booted up fine.
> 

> Conclusion: I believe there might be a limitation with respect to your firmware-util.sh script that doesn't allow a Linux install on a MicroSD card to boot when the MicroSD card is inserted into the MicroSD slot of the Chromebook.

the script has no such limitation. If you mean the RW_LEGACY firmware for your device might be buggy, then I would agree. The fact that the internal SD Card reader shows in the boot menu when media inserted means it **should** work. 

That said, he RW_LEGACY Firmware build for Apollolake devices is from 2018, it's quite possible that a newer build of SeaBIOS would resolve the issue, or switching to edk2/Tianocore. I'll see if I can make some time later today to try and reproduce here on REEF and test a newer build

closed 2026-11-04 by MrChromebox