MrChromebox/firmware #655: kefka 2in1 hinge not detected

Merith-TK, 2024-07-11:
The Kefka (mobile, not tying out its entire name) has a 2in1 tablet mode hinge, 
This special hinge is not visible to the installed OS, I have tried various Linux distros and kernels, windows 10 (did not work well due to no keyboard drivers), and recently Chromos Flex, none can detect the 2in1 hinge, it only shows as a regular laptop hinge.

I believe I am on latest firmware, updated it a few weeks ago, (don't have exact date)

MrChromebox, 2024-07-11:
Usually the convertible models have "360" appended to the HWID/board name, so possible we never came across a convertible KEFKA before -- will have to look into it.

Win10 should work fine, all drivers exist and work, can be found on coolstar.org / docs.chrultrabook.com

Merith-TK, 2024-07-11:
At my PC, here is the "official" name of it, 
"Dell Chromebook 11, 3189", is there anymore information I can provide that would be useful?

Merith-TK, 2024-07-11:
Okay further research shows that there are *two* kefka models, the 3180, which follows a standard laptop formfactor, and the 3189 which is the convertible one I have,

Merith-TK, 2024-07-12:
honestly I would be willing to beta test a custom build with hinge support, 
The only issue I would have is I have no way to recover from bricked firmware.

MrChromebox, 2024-07-12:
adding support for the accelerometer wouldn't brick the device by any means, I'll just have to see if the EC on kefka supports the method used by newer devices

Merith-TK, 2024-07-12:
Is there anything I can do to see if that is possible on my end?

MrChromebox, 2024-07-12:
the magnet is just the lid sensor.

The EC handles all the accelerometer related functions, it's just a matter of how that gets communicated to the OS (drivers) from the firmware.

Merith-TK, 2024-07-19:
I have learned.There's a very chroma that it's just a 3190.It is literally the exact same thing.My Guest is slightly better specs And just runs windows by default.So I can get my hands on one of that'll be real same model that the local school district has Given their students for summer school shit

Also.Sorry for rambling messages.I am using voice to text as i'm on a walk typing this

Merith-TK, 2024-09-16:
Any updates on this subject?

MrChromebox, 2024-09-16:
I can give you a test build with `#define EC_ENABLE_TBMC_DEVICE` set and see if that does anything, but if the actual EC firmware itself doesn't support the tablet mode switch (because it wasn't implemented yet) then it won't do any good.

Merith-TK, 2024-09-16:
I can try that, the latest firmware does support the gyro and things are detecting that it *should* be able to have a tablet mode (KDE is auto rotating, but the check for tablet mode is a no go)

Merith-TK, 2024-09-30:
any updates on that test firmware?

closed 2026-11-04 by MrChromebox