MrChromebox/firmware #656: SD was detected once, never again

null-VOID-null, 2024-07-12:
I have a dell Chromebook 11 3180 with a kefka motherboard, and I've recently installed the full rom uef, and made made a bootable SD but it only detected once when I had put ventoy on it and then put the iso of ubuntu mate on it, and it booted with no issues. But then I used the sd to Install ventoy on a usb as well as an iso like I did with the sd, I did this because I wanted to install Ubuntu make to the SD like you would a normal storage device. Nothing went wrong with the Installation process. When I rebooted and tried to boot the sd however, it just wasn't detected. All I saw was ubuntu (probably from my internal storage), and my internal storage. I don't know why it only detected it once and never again nor how to fix it.

MrChromebox, 2024-07-12:
the SD card reader on Braswell is directly connected to the mainboard, and so has to be inserted at boot to be detected - it won't autodetect on insertion like USBs will. Unfortunately there's no good way to debug edk2 on Braswell, since there's no serial output. One would need a device with a Google servo header attached, or to solder 0.5mm pitch wires to the solder pads to get serial output and see what's going on

null-VOID-null, 2024-07-12:
Ok then, well, I've done a bit if testing since I made this post, and it only really detects under these conditions:

Ventoy is installed on the SD

An iso file is on the SD

And it's the first boot after doing that. If I reboot it doesn't show up again.

So Im not sure if the issue is what you described, tho I could be wrong (and it is always inserted on boot btw) ((also, should I add the additional things I found to the post?))

MrChromebox, 2024-07-12:
does a warm vs cold boot make a difference in detection? "first boot' doesn't make much sense as a discriminating factor

null-VOID-null, 2024-07-12:
I've only tried cold booting, so idk. And I can't test it because I currently have no access to said SD (currently using it for other, more important tasks that would be wiped if I tried to test it)

closed 2026-11-04 by MrChromebox