MrChromebox/firmware #657: Wireless card issue on Pixelbook 2017

Gameparkwing, 2024-07-18:
The Google Pixelbook 2017 (EVE) is device I am using.
I had been upgraded the firmware to MrChromebox (verison 4.22.4 Release) and installed Windows 11.
The system is working normally, but the status of the wireless network card has a issue.
Wireless card can be working, but it appears in removable devices menu.
[image]

MrChromebox, 2024-07-18:
first I've heard of that on a Pixelbook, no idea why it would appear that way. There is nothing in the firmware programming to mark it as a removable device

MrChromebox, 2026-04-05:
@Gameparkwing use the latest release firmware and let me know if there's still an issue

closed 2026-09-27 by MrChromebox