MrChromebox/firmware #658: Fan runs at full speed after power loss Asus Chromebox 3 CN65 (teemo)

dantethemetaphor, 2024-07-24:
When the device (Asus Chromebox 3) reboots from power loss, the fans run at full speed. Only way I can get fans to run normally is to turn it off and boot manually from power button. I have noticed this behavior since updating firmware, (However I cannot recall which was the last firmware which didn't have this issue, my last update was somewhere last year)

MrChromebox, 2024-07-24:
can't replicate this behavior here on the 2405.0 firmware.

With OS booted, removing and reconnecting power results in a brief fan surge when powering on, as the EC is in RO mode, but as soon as it jumps to RW, the fan speed is returned to normal. 1-1.5s max.

dantethemetaphor, 2024-07-25:
Funny it appears to have stopped happening on one unit. Maybe because of clearing UEFI NVRAM, But I have another unit with a fresh install of Ubuntu and the same problem is happening, (running 2405.0 06/15/2024) . Clearing NVRAM on this unit doesn't resolve the problem

dantethemetaphor, 2024-07-26:
I have managed to run some more tests. The issue of full fan speed after power loss is indeed happening on all 3 of my units running 2405.0 (including the one I thought stopped exhibiting the issue). I have another unit with 4.20 which doesn't exhibit the behavior. I think I should also mention that I am able to get the fan to run normally and control it after booting into the os and running the chrultrabook-tool fan utility.  
Is there a way I could downgrade to 4.20 as we wait for a fix?

MrChromebox, 2024-07-26:
what mode is the fan in, per the chrultrabook utility (or ectool)? are the temps high?

I don't understand why I can't reproduce this here on either of my two units

please boot to the OS when this happens and grab a cbmem log and attach (not paste) here

dantethemetaphor, 2024-07-27:
Not sure if I got this right, i built cbmem and run cbmem -c 
[cbmem.log](https://github.com/user-attachments/files/16401558/cbmem.log)

dantethemetaphor, 2024-07-28:
the temps are normal, dont know how to check the fan mode, they fan runs at full speed right away so i assume they are running at max speed?, also as mentioned, if i reboot, or power off and turn on the unit using the power button the it boots normally. only when it auto powers up after disconnecting and reconnecting power do the fans run at max.

MrChromebox, 2024-07-28:
the cbmem log shows your device booting the EC in RO mode, then switching to RW. The fan speed should have corrected itself at that time. We can try setting it to Auto mode on boot and see if that helps

dantethemetaphor, 2024-08-09:
Hi did you ever get to the bottom of this? you mentioned setting the fan speed to auto at boot, will you be doing that in the next release or is there a way for me to do it on my end?

MrChromebox, 2024-08-10:
ping me via email or discord next week and I'll give you a test build w/auto enabled at boot, and we'll see if that solves the problem

magicalyu, 2024-09-01:
may I ask if this issue has been solved yet? I also have the same problem.

MrChromebox, 2024-09-01:
> may I ask if this issue has been solved yet?

I can't reproduce this on my units here. No one affected by the issue has offered to test a build with auto fan control enabled.

>  and seems the saved boot order is lost.

the bootorder is stored in UEFI NVRAM, on the flash chip. It is not affected by loss of power

magicalyu, 2024-09-02:
Thanks a lot for your reply.

> I can't reproduce this on my units here. No one affected by the issue has offered to test a build with auto fan control enabled.

I am very willing to help you test the build. Just tell me what should I do.

> the bootorder is stored in UEFI NVRAM, on the flash chip. It is not affected by loss of power

How can I test if the UEFI NVRAM is working or not?

Thanks again for your help

dantethemetaphor, 2024-09-02:
the issue is still occuring for me, unfortunately been abit busy so i had to have the unit flashed back to the older versions as they were running as home servers. i do have one extra unit and am willing to test the build with auto fan control. will send you an email @MrChromebox

magicalyu, 2024-09-02:
Hi, I also have sent a email to you @MrChromebox. 
very happy to help, if you still need.

magicalyu, 2024-09-02:
the fan issue does not appear anymore when flashed the new test build !
but the boot order will still be lost if power is lost.  
Thanks a lot!

MrChromebox, 2024-09-02:
> the fan issue does not appear anymore when flashed the new test build !

ok, will add this to the upcoming 2408 release then

> but the boot order will still be lost if power is lost.

this doesn't make sense to me, but I will investigate

dantethemetaphor, 2024-09-03:
problem appears to be solved with the test build, will keep testing to see if it comes back

closed 2026-11-04 by MrChromebox