MrChromebox/firmware #660: Cannot power Acer Chromebox CXI3 with MrChromebox firmware with Windows 11 with Power delivery (USB Type C)

ETG9466, 2024-07-28:
My specs are:
CPU: Intel Core i5-8250U
RAM: 8GB 2400MHz DDR4 RAM (SK Hynix)
Storage: 256GB SSD
Model: I58GKM

When I try to power my Acer Chromebox CXI3 with MrChromebox UEFI firmware with Power Delivery (the USB type C port), it gets stuck in a bootloop without the write protect screw installed, it gets a blinking red light with the write protect screw installed. But when I use the power brick, which when I move it around, it is very unstable and it goes off sometimes. And because of that problem, I want to power the Chromebox with Power Delivery (the USB type C port). Is there any way to fix it or can you (MrChromebox) please try to edit the UEFI ROM to fix the problem?

ETG9466, 2024-07-29:
Where was this GitHub issue transferred to?

ETG9466, 2024-07-29:
Is there any help to fix this issue?

ETG9466, 2024-07-29:
Any possible fix please?

MrChromebox, 2024-07-29:
dupe of #595, both devices use same baseboard, same EC firmware

closed 2024-07-29 by MrChromebox