MrChromebox/firmware #661: HP Chromebook 11A G6 EE turns on upon lid open or power cable plugged in even after full shutdown.

GamePlayer-8, 2024-07-29:
Good morning,

My Chromebook HP 11A G6 turns on upon opening the laptop lid, even if I've shut it down totally with `poweroff` command and I haven't pressed the power on button. It's also happening when plugging in the power cable.

How to turn it off?

ETG9466, 2024-07-29:
Easy answer, you can't. This happens on my school Chromebook when I log in to my school account on Chrome OS and I face that all the time. This is not a problem to me and you can't change what is happening with your Chromebook.

GamePlayer-8, 2024-07-29:
Meaning the lid as well as plugging in the cable sends the same signal as power button and Coreboot couldn't recognize the difference?

MrChromebox, 2024-07-29:
> My Chromebook HP 11A G6 turns on upon opening the laptop lid, even if I've shut it down totally with poweroff command and I haven't pressed the power on button.

this is standard Chromebook behavior. You can build coreboot and disable the lid opening as a wake source, but to disable it as a power-on source, you will need to build custom EC-RW firmware (same with booting on external power detect)

GamePlayer-8, 2024-07-29:
how to install built and tested ROM file onto the board with correct HWID and VPD?

MrChromebox, 2024-07-29:
I'm not sure what you mean "install built and tested ROM file"

closed 2026-11-04 by MrChromebox