MrChromebox/firmware #662: USB bakcup device failed to mount; cannot proceed.

TheGuyForMysecret, 2024-07-29:
SAME THING AS THIS IMAGE BUT THE NAME OF MY DRIVER IS "easystore_2647" 

sudo crossystem hwid = OMNIGUL-FUQB D4B-D4E-16C-G36-J6A-A6P
sudo dmidecode -s system-product-name = Omnigul

[image]

MrChromebox, 2024-07-29:
the script prompt says:

> Enter the number for the device to be used for the firmware backup:

you're supposed to enter '1'

TheGuyForMysecret, 2024-07-29:
> the script prompt says:
> 
> > Enter the number for the device to be used for the firmware backup:
> 
> you're supposed to enter '1'

Oh my bad brother I feel like a retard rn. Apologies

TheGuyForMysecret, 2024-07-29:
> the script prompt says:
> 
> > Enter the number for the device to be used for the firmware backup:
> 
> you're supposed to enter '1'

Still says the same thing for me.

[image]

MrChromebox, 2024-07-30:
as per the docs, the USB backup drive needs to be FAT32-formatted - the external USB HDD/SSD you have connected is almost certainly not. Do you not have a spare USB stick you can format and use?

TheGuyForMysecret, 2024-07-30:
> as per the docs, the USB backup drive needs to be FAT32-formatted - the external USB HDD/SSD you have connected is almost certainly not. Do you not have a spare USB stick you can format and use?

My Drive is a FAT32

[image]

MrChromebox, 2024-07-30:
you have 2 drives plugged in? If not, you're formatting the 2nd partition, and the script can only use the first

closed 2026-11-04 by MrChromebox