MrChromebox/firmware #663: UEFI Menu & Coreboot logo during boot shows only when USB C (Display port) monitor is connected otherwise blank

bmltp, 2024-08-08:
Device is 
** Device: Lenovo Chromebook S340 / IdeaPad 3
** Board Name: LASER14
**   Platform: Intel GeminiLake
**    Fw Type: Full ROM / UEFI
**     Fw Ver: MrChromebox-2405.0 (06/15/2024)
**      Fw WP: Disabled

After successful installation & during reboot screen seems just black with just light (pressing ESC & reinstalling not fixing the issue). however if I connect USB C monitor then Coreboot logo & UEFI/BIOS menu appears. During first reboot it seems just black screen with light because it is showing no bootable device found which can be seen only after rebooting with connected USB C monitor but display appears on laptop screen. After installing OS my case Debian. it works (display appears on laptop screen) normally once Linux loads its stuff.

MrChromebox, 2024-08-08:
no idea why that would be, it uses the same display config as all other GLK ChromeOS devices. Will need a cbmem console log to see if there's anything useful in there

bmltp, 2024-08-12:
I have attached the cbmem console log. Please have a look. Thank you.
[cbmem.log](https://github.com/user-attachments/files/16577185/cbmem.log)

MrChromebox, 2024-08-12:
everything looks good in the log, all I can think is that the VBT in my firmware is older than what ChromeOS ships and doesn't support the display panel used in your device. We can try a test build with an updated VBT -- drop me an email or ping me on Discord/IRC

bmltp, 2024-08-13:
I installed the test build and now it work as expected. Thanks a lot.

closed 2024-08-13 by bmltp