MrChromebox/firmware #664: M.2 Wifi card slot dosn't work on BUDDY

Zema4You, 2024-08-10:
Generally Buddy works fine with MsChromebox firmware, thank You😊
I’m trying to install Realtek 1GbE card (RTL8111) in Buddy Wifi M.2 slot but card is not active in OS and is not listed in UEFI Shell by PCI command.
I have modified my card to give GND to CLKREQ# pin and card started working in other mainboards (thank two [namezys](https://github.com/namezys) in https://github.com/MrChromebox/firmware/issues/512#issuecomment-1703398317 ), but not in Buddy.
I have tried also another 2 wifi cards in this slot but none worked.
Any suggestions how to start wifi slot working in Buddy?

Have a good day

MrChromebox, 2024-08-10:
only thing I can think of is to disable CLKREQ usage on the wifi port and see if that helps - would require a custom firmware build

Zema4You, 2024-08-10:
Let’s try this. It can help if CLKREQ is not connected to the M.2 socket at all.

MrChromebox, 2024-08-12:
drop me an email or ping me on IRC/Discord and I'l give you a test build to try

Zema4You, 2024-08-16:
I have found one card that works in M.2 Wifi slot in BUDDY - Intel I210. 
Then the issue is limited only to some cards.
I will investigate further and give info when find something new.

Zema4You, 2024-08-17:
I have rechecked 5 cards and all that have connected CLKREQ, also Realtek card after grounding CLKREQ works (again thanks two [namezys](https://github.com/namezys) in https://github.com/MrChromebox/firmware/issues/512#issuecomment-1703398317 ).

So it was human failure when I was checking before and issue is false positive.

Best regards

closed 2024-08-17 by Zema4You

wicadmin, 2024-08-17:
Hey @Zema4You - Followed up a question here: https://github.com/MrChromebox/firmware/issues/512#issuecomment-2294882030

Thank you.