MrChromebox/firmware #665: CVE-2023-31315 sinkclose

DSDV, 2024-08-11:
Hello, 

there was a recent AMD CPU security flaw: 
https://www.amd.com/en/resources/product-security/bulletin/amd-sb-7014.html

According to AMD some of the Chromebook CPUs like the 3000 Series got AGESA Updates.
For Example:
```
Picasso-FP5
1.0.1.2
(2024-08-06)
```

is it possible that Coreboot updates will fix this as well? 
Or do we have to revert to ChromeOS for a fix and after reflash coreboot? 

with kind regards

MrChromebox, 2024-08-12:
this issue is not applicable to Chromebooks, or any devices running coreboot -- the CVE is with the UEFI SMM mode access, which does not exist on Chromebooks.

closed 2024-08-12 by MrChromebox

n3k, 2025-01-25:
This is not true and Coreboot is obviously not a solution to the issue.

MrChromebox, 2025-01-25:
> This is not true and Coreboot is obviously not a solution to the issue.

if you're going to drive by and shitpost, at least do us the courtesy of a explanation, assuming you have one

n3k, 2025-01-25:
Define shitposting?... The vuln happens at a deeper level than what Coreboot can handle. SMM cannot be disabled as an execution mode. While Coreboot might not extensively use it, the execution mode still exists and the vulnerability could be used to get arbitrary code execution in this mode. Check this out: https://github.com/coreboot/coreboot/blob/main/src/cpu/x86/smm/smm_stub.S#L51 
That's a mitigation against a similar condition that affected Intel Sandybridge and before. A patch could be made to check for the TClose bit upon the SMI entry occurs and bail out in the same manner for the AMD case. This would prevent exploitation in case the microcode updates from AMD are not deployed or not available for whatever reason. This is also mentioned in the details of the publication about the Sinkclose attack btw.

MrChromebox, 2025-01-26:
> Define shitposting?

Writing "this is not true" (what is 'this' anyway? you didn't quote anyone, so it could be anything prior) and "coreboot is obviously not a solution" (a claim which no one made) without any explanation

As I'm not an AMD employee anymore, I don't have visibility into the fixes for this CVE, so have reached out to some former colleagues to check and see if any AMD ChromeOS products (running coreboot) were deemed vulnerable and if so, what the mitigations are.

doritos4mlady, 2025-01-27:
> As I'm not an AMD employee anymore, I don't have visibility into the fixes for this CVE, so have reached out to some former colleagues to check and see if any AMD ChromeOS products (running coreboot) were deemed vulnerable and if so, what the mitigations are.

It would also be good to know if excavators and earlier are affected by this vulnerability. 
In the defcon presentation Nissim & Okupski spent a lot of time talking about SMT and they only tested it on SMT systems. Perhaps CMT systems are immune?