MrChromebox/firmware #667: > > FWIW, you may install EndeavourOS or another Arch-based distro and have working Bluetooth and USB sound using the audio script linked in the docs.

PD-Kerman, 2024-08-18:
> > FWIW, you may install EndeavourOS or another Arch-based distro and have working Bluetooth and USB sound using the audio script linked in the docs.
> 
> Funny you should say that, because I just blew away lubuntu and installed Manjaro. I had to add a couple of ufs modules and mkinitcpio. Works like a champ. So, I'll dig out the script and take a shot at it.
> 
> Thanks again!

Hi there
I just got a elemi / hp c640 g2 here and seems like the issue still remains on this model, which it doesnt seem like i could boot the system from the microsd card in the slot~ 

ill first paste the brief post on discord (chrultraboook) but unfortunately there r no support on the discord server~
[image]

# So after reading all previous comments/discussions under this thread, i guess ill just firstly post these infos cuz they should be related most: 

-  **FW_VER: Okey so firstly, i just ran the installation script today, and the date of the firmware seems to be 06/15/2024 as its been shown here** 

[image]

- **BOOTMENU: And according to the previous discussion i noticed that i the device should still being displayed in the boot menu regardless of whether its bootable so the following attachment show that the microsd card was nicely plugged in and i can confirm that it was recognized in all installation media and its been tested usable**
[image]

[image]

- **MISCELLANEOUS: Here's some additional screenshots about the blank file explorer and the UEFI Shell with `map: No mapping found.` notice which seems like no devices were even detected even tho the microsdcard had been inserted**

# What i did: 

**Well i was originally wanted to install the entire operating system on the microsd card in the built in slot cuz the slot could get it locked inside which should be save enough and my internal storage was like i only got 64GB which wasnt enought and speaking of which _the internal eMMC didnt displayed on the firmware as well_

and i _tried both ubuntu and arch linux and the USB installation media works both perfectectly fine_, and ive attempted to install it to both eMMC and microsd card separately _both didnt boot_ (*might need to get internal eMMC one tested again to confirm)

_which means yea even tho the microsdcard is bootable/freshly installed, it is still not detectable_

and about the microsd card its a sandisk extreme 256GB and again works perfectly fine ive been tested it for several times~
**

its kind of a really important feature and i really wish it could be fixed
please let me know if any other infos were required :D

All kinds of help at all would be very much appreciated!~ 
Thanks in advance

_Originally posted by @PD-Kerman in https://github.com/MrChromebox/firmware/issues/572#issuecomment-2295226186_

PD-Kerman, 2024-08-18:
### _**Attaching cbmem.log, dumped using ubuntu live booted from usb flash drive cuz due to the non-boot-able issue of both emmc and microsd we cant really get into the os and dump logs in that kind of scenario for it, hope it would still be useful**_
[cbmem.log](https://github.com/user-attachments/files/16648357/cbmem.log)

PD-Kerman, 2024-08-19:
### Also posted on chrultrabook forum with a reformatted and briefer description regarding the issue 
https://forum.chrultrabook.com/t/elemi-issue-with-uefi-full-rom-emmc-microsd-non-bootable/3042

MrChromebox, 2024-08-19:
ELEMI uses a PCIe-->eMMC bridge, which coreboot properly initializes. So it seems the problem is with edk2 not detecting it. The easiest solution is to swap the eMMC board for a NVMe one, assuming it's socketed. Otherwise, will need to flash a debug build with edk2 logging and log via a SuzyQ cable

PD-Kerman, 2024-08-19:
> ELEMI uses a PCIe-->eMMC bridge, which coreboot properly initializes. So it seems the problem is with edk2 not detecting it. The easiest solution is to swap the eMMC board for a NVMe one, assuming it's socketed. Otherwise, will need to flash a debug build with edk2 logging and log via a SuzyQ cable

Thank you for your reply, 

Well but I mean like some other device could boot from it properly~ so is it because ELEMI uses that kind of bridge that made it different or the other device wasn't ought to be able to boot via eMMC as well~?

And is or will there be a new firmware-release/simple that could fix this issue and let edk2 got it detected~? 
Thanks

Best regards,
PD-Kerman

MrChromebox, 2024-08-19:
> Well but I mean like some other device could boot from it properly right?~ [for example seems like other volteer devices could boot from it as well right~ according to the discord discussion, or is it just ELEMI was an exception~? :D] so is it because ELEMI uses that kind of bridge that made it different or the other devices wasn't ought to be able to boot via eMMC as well~?

I have no idea. I only have a single TGL device with NVMe and it has no issues. I think most volteer devices use NVMe, not eMMC. I'm not sure if other volteer devices use eMMC, I'd have to check.

> And is or will there be a new firmware-release/simple-fix that could get this issue fixed and let edk2 got it detected~?
Thanks

without an edk2 debug log, I have no idea what the issue is or how to potentially fix it

PD-Kerman, 2024-08-19:
> I have no idea. I only have a single TGL device with NVMe and it has no issues. I think most volteer devices use NVMe, not eMMC. I'm not sure if other volteer devices use eMMC, I'd have to [...]

Okey i get that now, so I'll have to get a debug build of edk2 firmware flashed into it in order to get a better figure of what went wrong right~? :D Well that indeed cleared things up~ 

Although I don't personally got a SuzyQ here but I've got a brief review from a contributor/moderator from chrultrabook discord server: general channel mentioned that he's got several volteer devices and I think he did mentioned that some of them have eMMC storage so, if some of his volteer devices had indeed got eMMC storages then is it possible that he could run the test and get the logs from some volteer devices as reference in case of that it's got the same non-bootable eMMC storage issue as well~?

Again thanks for your reply
Best regards,
Pd-Kerman

PD-Kerman, 2024-08-19:
Attatching the detailed conversation log in the discord server
P.S. Jamie had got some point of view regarding RP9 in the device tree of ELEMI in this case might need your review as well

Thanks in advance

[image]

mcirsta, 2025-11-17:
Right now, with the latest Elemi firmware and latest full UEFI it all works fine with CachyOS and grub but Windows11 throws all kinds of errors, sometimes doesn't even see the eMMC and sometimes it gets to 75% but still fails when reading/writing from the eMMC.
 Not sure if it's a Windows driver issue or just how the eMMC is initialized.

MrChromebox, 2025-11-17:
cbmem log please

https://docs.mrchromebox.tech/docs/support/debugging.html#firmware

mcirsta, 2025-11-17:
Sure thing, I've attached it. I think I might keep Linux anyway as it seems to work nicely on this laptop thanks to the full UEFI BIOS but still for people trying to use Windows 11 it might be nice if there's a fix for it. 

 I am willing to test but I will probably lend the laptop for 1-2 month. I will test when I get it back for sure. I've tried with the latest Windows 11 ISO from the MS website.

[cbmem.log](https://github.com/user-attachments/files/23587327/cbmem.log)