MrChromebox/firmware #669: Chromebox3 can't Auto poweron when AC Recovery

InfinityWang, 2024-08-22:
Hello,my Chromebox3(i7-8550u) can't auto poweron when AC recovery.When I give power, I only see the red light and immediately turn it off. I manually press the power button to keep the white light on and start. But it can automatically restore when forcibly powered off and then powered on again. Can I set auto power on when AC recovery in any situation? Is this because my BIOS version is too old? If so, how should I update it? It is currently running window11.

MrChromebox, 2024-08-22:
the default behavior is to return to the state when AC power was lost -- so if the device was on, it will power back on, but if the device was off or in suspend/sleep, it will stay off.

Custom firmware is required to automatically power on regardless of previous power state

InfinityWang, 2024-08-22:
Thank you very much for your reply! I will try!

MrChromebox, 2024-08-22:
I'll look into adding an auto-on build for the next release

InfinityWang, 2024-08-22:
Thank you for your support!!!

InfinityWang, 2024-08-22:
This device is still very popular on the Internet.

MrChromebox, 2024-08-22:
I know, I have 2 or 3 of them myself ;-)

InfinityWang, 2024-08-22:
Wow, that sounds fucking cool 🤤

closed 2026-11-04 by MrChromebox