MrChromebox/firmware #677: ChromeOS won't boot and recovery media doesn't work

animmjfan, 2024-09-20:
The ChromeBook is a vilboz and I used recovery media to flash the rom back to the vilboz chromeos firmware using recovery media and the script to revert but when I used recovery media to install chromeos it failed I'm writing this from the chromebook by using shimboot and I disconnected the battery to be able to attempt  to flash full rom and try again but flashing full rom requires making a usb back of the rom but that doesn't work because I have a not completely normal eMMc  as it doesn't have full CrOs installed and I have no way to progress I tried to disable wp but flashrom doesn't work as the chromebook it booted into debian with cinnamon via shimboot and just using shimboot to use something like brunch or chromeos flex won't work as that requires making patches to systemd in chromeos which I can't boot into anyways so my idea is to get this into the full rom and then go back and see if it fixes it if not then I'll just have to either find a rom somewhere, have vilboz get added to the net recovery method of flashing chromeos rom or use brunch but none of that can work without finding a way pass the make a rom backup part of flashing full rom

pinpolhoxx, 2024-09-20:
This might help:This file might fix it
https://bit.ly/3TzaAQd
Pass: changeme
I put the necessary dlls in the archive

animmjfan, 2024-09-20:
it looks like it's passwd locked and also looks like it's for windows?

animmjfan, 2024-09-20:
I can only boot debian through a tool known as shimboot and dev mode

animmjfan, 2024-09-20:
actually I think something went wrong on the usb I was using a usb that had the logs of the recovery error to backup the rom too but I plugged the drive into windows and used diskpart the clean the drive and tried again but the script didn't like the drive and said could not mount so a opened "Disks" and made a partition and it worked

animmjfan, 2024-09-20:
just keeping open until my original issue is fixed just incase

animmjfan, 2024-09-20:
I'm currently in full rom but i don't know if this will work

MrChromebox, 2024-09-20:
I can't tell from that wall of text what your issue actually is

animmjfan, 2024-09-21:
I think I need a shell bash image for my Vilboz ChromeBook

MrChromebox, 2024-09-21:
Assuming you mean a shellball image, that's what you flashed when you used the recovery media to restore the stock firmware via my script

animmjfan, 2024-09-22:
So do I just flash using recovery media?

MrChromebox, 2024-09-22:
yes

animmjfan, 2024-09-23:
Okay will try!

animmjfan, 2024-09-28:
UPDATE:
I got the chromebook back into chromos and disconnected the battery then recovered which worked but now after connecting the battery to the board the battery is working and the chromebook will boot off of the battery but I have a red flashing light recovery mode says the battery level is 0% and the Chromebook has removed the Battery from the quick settings and diagnostics is there a way to fix because I don't want to have to hold power for a long time just to boot or the pain of trying to enter recovery mode in this state

MrChromebox, 2024-09-28:
your symptoms are consistent with the battery not being connected/connected properly

animmjfan, 2024-09-30:
But the battery is working without being plugged in the laptop still boots and runs?

animmjfan, 2024-11-04:
sorry for late update! it turns out that since this device is used with the school (Note. NOT OWNED BY SCHOOL I HAD TO PAY FOR IT) the IT teacher who is fully okay and supportive of doing things to it as long as it's still usable. They let me give it to someone to check (note. at the time I didn't get enough time to even try taking it apart quickly because school started and multiple times! and I got distracted by other projects) and after a few days I was given it and told that somehow the battery was non working and was replaced (Note. The reason I even did send it in was because I didn't get any time, and the school does repair fully free. so no loss because they are supportive) my guess of what happened is that because I was flashing a lot, the battery somehow got damaged? it's perfectly fine now and I actually was only sending replies and updates to this thread using the device, it started in Linux through either full rom or shimboot (a rma shim exploit) and ChromeOS while the device battery still had power (after power ran out it wouldn't charge) and after it was fixed. Thank you for all the help!! sry for all the "walls of text"! hope you have a good day!

closed 2024-11-04 by animmjfan