MrChromebox/firmware #680: Intermittent Bricking - Lenovo IP Flex 5 Chrome 14IAU7

gdbaldw, 2024-09-26:
All installed and working perfectly, running Debian Testing on a 2TB internal card. But, randomly the machine will not boot, no keyboard backlight, nothing. Even the technician at the Lenovo Service Center is stumped. Sometimes, if I disconnect the battery and the keyboard ZIF cable, it will boot on the power supply. But, other times it boots ok with battery and keyboard cable connected. The technician is suspecting a hardware problem, but cannot isolate any component issue. But, with a mrchromebox bios, I suspect that maybe its some sort  of suspend or hibernate issue. I really don't know where to begin to debug the issue.

MrChromebox, 2024-09-26:
does a hard reset (Refresh+Power) revive the device when it won't boot?

gdbaldw, 2024-09-26:
Refresh? Don't know what/where on this model. Is it a key combination? I did try the key combination to put into maintenance mode, but did not boot.

MrChromebox, 2024-09-26:
it's the circular arrow, typically where F3 would be

[image]

I don't know what you mean by maintenance mode, there is no such thing on Chromebooks. Do you mean Recovery Mode (ESC+Refresh+Power)?

gdbaldw, 2024-09-26:
I had tried <esc><circle arrow><power>, but did not work. Next time it bricks, I will try <circle arrow><power>.

 
 

image.png (view on web)

marcosilva85, 2024-11-08:
@MrChromebox I report the same soft-brick at boot, present in latest 3 releases (from 2408 of MrCB) on Lillipup and I happened to capture a screenshot of this issue:
[image]

on boot, after the rabbit logo, Screenshot presents:
"SecureBoot status: Disabled
Booting from 'Garuda' failed: verify it contains a 64-bit UEFI OS.
Press any key to continue booting..."

All attempts to try to boot from here fail. Pressing any key goes to "UEFI Interactive shell v2.2. EDK II. UEFI 2.4" and there's a line saying "map: no mapping found". But typing "bcfg boot dump -b" lists all my entries nicely though.

Seems not easy to guess when it decides to soft-brick, as sometimes is from restart, sometimes seems from booting by opening the lid.
To get past it, I attempt many shutdowns (no restart, cold boots) till luck gets you to the normal grub loader instead. Somehow I feel if I wait minute(s) before attempting a cold boot works to pass around the problem. 

Sadly we can't rollback with your script, from what I understand it's what you prefer for avoiding entropy with too many issues, which is OK! Nevertheless we hope you can find if it can be fixed and wish its an easy thing! :+1:

MrChromebox, 2024-11-08:
@marcosilva85 when it dumps you to the shell, type exit and press enter as per the instructions. then go to the boot menu. Is the internal storage listed as a boot device?

marcosilva85, 2024-11-08:
@MrChromebox no, the NVMe does not show in there.

Usually it lists 4 entries with NVMe at bottom, but with this issue active it only shows "Garuda, Fedora, UEFI Shell" (and sadly the first ones then do not boot, only UEFI shell )

MrChromebox, 2024-11-08:
then that's the issue, the NVMe isn't coming up on a warm boot sometimes

marcosilva85, 2024-11-08:
yeah can be that, sadly!
but I also even try to restart the laptop from that state, expecting the NVMe to show up, but the issue just persists...  its difficult to come up with a quick solution to boot when it gets into this state, regrettably

I also just came across #671 which seems it might be present in different devices,
Any way you could figure out a fix for this ?

marcosilva85, 2024-11-08:
@MrChromebox  yeah can be that, sadly!
but I also even try to restart the laptop from that state, expecting the NVMe to show up, but the issue just persists...  its difficult to come up with a quick solution to boot when it gets into this state, regrettably

I also just came across #671 which seems it might be present in different devices,
Any way you could figure out a fix for this ?

MrChromebox, 2024-11-08:
@marcosilva85 your issue seems unrelated to this ticket, and #671 is for a device with UFS storage, not NVMe. Please open a separate one for LILLIPUP

gdbaldw, 2024-11-09:
I have kept my laptop running with power to avoid the issue. However, I can offer an anecdote that may be helpful. Each time my laptop bricked, I took it into the local Lenovo service center, and each time it "miraculously" booted. I don't know why. I would try for an hour at home, no luck. But, at the center it seemed to boot after only a few tries, within 5 minutes. The first time this happened, the bootup process froze and the last message in the bootup sequence said something about Bluetooth, if I recall correctly. I connected to a power outlet at the service center, but sometimes just the battery. All this suggest to me anyway that maybe there was some WiFi or Bluetooth signal at the Lenovo service center that was helpful to the boot process. I asked the service center tech if they were running boot on WiFi or some such signal that might affect my laptop, and he said no, not that he was aware of. Sorry I cannot be of further assistance, as I will just continue to stay powered on until we have a fix.

MrChromebox, 2024-11-09:
@gdbaldw what board are we even dealing with here? which firmware type?

gdbaldw, 2024-11-12:
ChromeOS Device Firmware Utility Script [2024-11-01] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**     Device: Lenovo Lenovo Flex 5i Chromebook 14"
** Board Name: TAEKO
**   Platform: Intel AlderLake/RaptorLake-U/P
**    Fw Type: Full ROM / UEFI
**     Fw Ver: MrChromebox-2408.1 (09/14/2024)
**      Fw WP: Disabled

I replaced the stock 128GB NVMe with a 1TB NVM3 2230.

MrChromebox, 2024-11-12:
do you have a SuzyQ cable available? That's really the only way to debug an intermittent booting issue

gdbaldw, 2024-11-13:
> do you have a SuzyQ cable available? That's really the only way to debug an intermittent booting issue

I have. Yes.

MrChromebox, 2024-11-13:
unzip and flash this via SuzyQ: `sudo flashrom -p raiden_debug_spi:target=AP -w coreboot_edk2-taeko-mrchromebox_20241113.rom --fmap -i COREBOOT -N`

then log to file using minicom via /dev/ttyUSB1. one log of the working case and one non-working would be ideal.

[coreboot_edk2-taeko-mrchromebox_20241113.zip](https://github.com/user-attachments/files/17737018/coreboot_edk2-taeko-mrchromebox_20241113.zip)

gdbaldw, 2025-02-12:
Sorry for the delay. I finally have another Linux machine, and of course I now urgently want to fix this.

Successfully executed...

`doug@debian:~/Downloads$ sudo flashrom -p raiden_debug_spi:target=AP -w coreboot_edk2-taeko-mrchromebox_20241113.rom --fmap -i COREBOOT -N
flashrom 1.4.0 on Linux 6.12.12-amd64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

FISK: AP
Raiden target: 2,0
Found XMC flash chip "XM25QH256C" (32768 kB, SPI) on raiden_debug_spi.
===
This flash part has status UNTESTED for operations: ERASE WRITE WP
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
You can also try to follow the instructions here:
https://www.flashrom.org/contrib_howtos/how_to_mark_chip_tested.html
Thanks for your help!
Using region: "COREBOOT".
Reading old flash chip contents... done.
Region [0x01c04200 - 0x01ffffff] is not sector aligned! Extending start boundaries by 0x00000200 bytes, from 0x01c04200 -> 0x01c04000
Erase/write done from 1c04000 to 1ffffff
Verifying flash... VERIFIED.
doug@debian:~/Downloads$ 

I don't know minicom, but tried it. CTRL-A L, but the log file had no contents. Status bar says OFFLINE.

Using `sudo screen /dev/ttyUSB0 115200
` I see
`[2375.909465 AP UART on]
[2376.233314 AP UART off]
[2377.239356 power button pressed]
[2377.275363 AC: -F]
[2377.284301 AP UART on]
[2377.325038 AC: R-]
[2377.593883 AP UART off]
[2377.624683 power button pressed]
[2377.662796 AP UART on]
[2377.962532 power button pressed]
[2377.992916 AP UART off]
[2378.000610 AP UART on]
[2378.288463 power button pressed]
[2378.589206 power button pressed]
[2378.956779 AP UART off]
[2379.989921 AC: -F]
[2380.038127 AC: R-]
[2612.229701 boot_mode: 0x01 -> 0x00]
[2612.230888 enable_spi_pinmux: AP]
[2637.563854 usb_spi_board_disable]
[2637.961872 boot_mode: 0x00 -> 0x01]
[2638.571010 AC: -F]
[2638.596424 AP UART on]
[2638.631507 AC: R-]
[2638.933528 AP UART off]`

Using `sudo screen /dev/ttyUSB1 115200` I see nothing

minicom to USB0 shows this...

[0.068065 Console unlock allowed]
[0.068885 tpm_rst_asserted]
[0.069389 boot_mode: 0x00 -> 0x01]
[0.070552 EC-COMM: Initialization]
[0.071086 FIPS module digest a4459b5bcbda8970...]
[0.071754 Running in FIPS 140-2 approved mode]
Console is enabled; type HELP for help.
[0.073986 CCD state:]
[0.074531 EC RX only]
[1.068862 AP UART off]
[1.074279 Servo disconnect]                                        
[1.075191 EC on]                                                   
[3.069507 AP off]                                                  
[3.107234 Rdd connect]                                             
[4.075141 CCD EXT enable]
[4.075984 USB PHY B]
[4.076965 CCD state: UARTEC+TX I2C SPI USBEC+TX]
[4.078159 I2C connect]
[4.078685 Initialized I2C port 0, freq = 100 kHz]
[4.161716 usb_reset, status 4801020]
[4.224895 usb_reset, status 9028]
[4.556850 usb_reset, status 801020]
[4.621082 usb_reset, status 9028]
[4.883674 SETAD 0x0b (11)]

but logging shows an empty file.

Please advise...

spawts, 2025-02-23:
Here is my USB1 minicom capture.

https://drive.google.com/file/d/1AI9WOndcWYA8Mv6NwLocZ0OzxpvMfz8l/view?usp=sharing

closed 2026-11-04 by MrChromebox