MrChromebox/firmware #682: Morphius/lenovo C13: No keyboard backlight

skuperuser, 2024-10-01:
Hi there, the keyboard backlight on my Lenovo Thinkpad C13 yoga seems to be broken.

I've tried turning it on with windows & crosec shortcut and on fedora linux KDE. Linux/KDE does recognize that the laptop has a keyboard backlight, but changing the keyboard lightning slider does nothing. On top of that, the keyboard does not light up in the BIOS, if that matters.

So, I think this is most likely a firmware issue. The lighting was working just fine on rw_legacy.

My model:
ryzen 3500C, 8gb ram, 128gb nvme

Are there any fixes for this?

MrChromebox, 2024-10-01:
the keyboard backlight is working fine here on my C13 with 2408.1 firmware PopOS. Sounds like an OS/driver issue to me

skuperuser, 2024-10-01:
Tried a few things like setting the brightness in console. i've also updated the uefi 2024.09.08>2024.09.14, cleared nvram, ran mint and popos, but it's still broken.
[image]
[image]
[image]

i've also found some others reporting this issue on the chrultrabook forums
https://forum.chrultrabook.com/t/lenovo-thinkpad-yoga-c13-gen-1/420
https://forum.chrultrabook.com/t/thinkpad-c13-ryzen-5-linux-support/2894/7 (last comment)

maybe it could be specific to this hardware config?

MrChromebox, 2024-10-01:
it's the same hardware config as I have here.

Setting the brightness value to 100 is less than half, the scale is 0-255.

The firmware has done its job if the chromeos kb backlight driver has attached and the controls are visible via sysfs.

cadecomposer, 2024-11-14:
Hi, figure I should chime in and mention I've got the same issue on my C13. It was working fine for a while but it suddenly stopped for some reason. When I use ectool it still reports that the keyboard backlight is at 100, and I'm able to change it directly through the GNOME QS panel and the changes are reflected in ectool, so I'm not sure why the backlight doesn't light up anymore. I wonder if it's some other issue unrelated to the firmware. If I get time tomorrow I'll try booting to a different distro from USB to see if it's any better, or maybe I'll have a dig around and see if it's a hardware issue (hopefully not...)
[image]

skuperuser, 2024-12-15:
Hi again, thanks for getting back to me. I've just forgot about it for a while, until I decided to restore chromeos (and reconnected keyboard connectors 10 and 11), but it still doesn't work. It did work before i first flashed the bios, so i'm not sure what it is, but it's probably not the firmware.