MrChromebox/firmware #688: Warning message

Castaevsky, 2024-10-20:
Hello Mr Chromebox,
I would like to ask you about the following issue: I had previously installed your firmware on my Acer Chromebook CB314 (Droid), but then decided to revert back to the stock firmware. However, after doing so, I am now getting a message at the first boot after recovery. Is this a critical issue? Can it be solved? Do I need to do anything about it?
Thank you for your help.
[image]
[image]

MrChromebox, 2024-10-20:
did you restore the stock firmware using a backup made by the script on USB, or using a recovery USB, so something else?

what is the current HWID?

Castaevsky, 2024-10-21:
"did you restore the stock firmware using a backup made by the script on USB, or using a recovery USB, so something else?
"
Initially, I used a recovery USB created by a Recovery tool when the battery was disconnected during the flashing process. However, I received a warning and decided to try again. The second time, I used an image from Cros.tech and the DD command on Linux while the battery was plugged in during the flashing process.
"
did you restore the stock firmware using a backup made by the script on USB, or using a recovery USB, so something else?

"what is the current HWID?"
My current hardware ID is "DROID ABC-123-XYZ-456".

MrChromebox, 2024-10-21:
> Initially, I used a recovery USB created by a Recovery tool when the battery was disconnected during the flashing process. However, I received a warning and decided to try again. The second time, I used an image from Cros.tech and the DD command on Linux, with the script on USB, while the battery was plugged in during the flashing process.

I was asking how you restored the stock firmware via my script, not how you restored ChromeOS

> "what is the current HWID?" My current hardware ID is "DROID ABC-123-XYZ-456".

that tells me you restored the stock firmware using a recovery USB and not your own backup. The firmware on the recovery USB doesn't contain a valid HWID, serial, or anything else specific to your device. This is why restoring from your own backup is strongly preferred.

All you can do at this point is change the HWID to something valid. if you do still have your stock firmware backup on USB, it can be extracted rom that, otherwise you'll have to find one someone shared somewhere.

Castaevsky, 2024-10-21:
Thank's!

closed 2024-10-21 by Castaevsky