MrChromebox/firmware #689: Firmware bricked (hp chromebook 11 G7 EE)

GreenyAU, 2024-10-23:
done the conversion a few times on the same model...no issues. this time i followed the same steps on the website, did everything correctly (double checked each step at a time), it installs the new firmware and goes back to its main command screen, i reboot with the 'r' command after roughly 2 mins on the main screen, it turns off and never comes back to life. im confused af right now. thanks chromebook, im stuck on a mac thatll only accept linux os's (2011 imac, ive tried everything...windows 7 cd is my last resort before some pc shops gets to tinker) (i just want my games. sad face.) 

also attached the stock firmware rom thing i backed up from the chromebox command tool... if its somehow useful. 
[stock-firmware-MIMROCK-20241023.rom.zip](https://github.com/user-attachments/files/17490583/stock-firmware-MIMROCK-20241023.rom.zip)

MrChromebox, 2024-10-23:
https://docs.mrchromebox.tech/docs/support/unbricking/

closed 2024-10-23 by MrChromebox

GreenyAU, 2024-10-24:
> https://docs.mrchromebox.tech/docs/support/unbricking/

Oh my, well at least I get to learn more stuff