MrChromebox/firmware #691: USB port drivers missing in Full ROM (UEFI) Firmware on KARMA devices

njalooo, 2024-10-24:
Hello everyone, i successfully installed the Full ROM Firmware on the Acer Chromebase CA24V2, by adding the write protect override screw. However there is one big issue for proceeding easily, and that is that the main USB ports directly on the Mainboard are not working with the new firmware. 
Luckily, the one Port and SD slot on the side extension were working, so i was able to load the installer on an SD card, and connect the Keyboard via USB.
After installing Debian 12, the ports worked again.
It would be very inconvenient however, if one did not have a SD card at hand.

MrChromebox, 2024-10-24:
cbmem and dmesg logs please, as well as the output of `pci -nn` and `lsusb -v`

https://docs.mrchromebox.tech/docs/support/debugging.html

njalooo, 2024-10-25:
Sure, here you go. I ran the debugging script, however the script said it was unable to find dmidecode, even though i had the package installed.
[debug-logs-Karma-2024-10-25_13h29m.tar.gz](https://github.com/user-attachments/files/17521346/debug-logs-Karma-2024-10-25_13h29m.tar.gz)

MrChromebox, 2024-10-25:
nothing in your logs shows any USB related errors, and all ports appear correctly initialized by the firmware.

If you connect a USB device to one of the non-functional ports after booting the OS, is there any error msg in dmesg or other system log?

njalooo, 2024-10-27:
No, nothing new shows up in dmesg when i plug something in. However, when i plug it in the power LED of the USB stick goes on for about half an second, before loosing power again, meaning the USB port stops delivering power. This happens every time i re-plug.

Here are the logs anyways; and i enabled secureboot in the meantime, so that could explain some differences.
[debug-logs-Karma-2024-10-27_12h26m.tar.gz](https://github.com/user-attachments/files/17533638/debug-logs-Karma-2024-10-27_12h26m.tar.gz)

I have disabled write protect via the screw-head, which is still in. Could that have an effect?

MrChromebox, 2024-10-27:
> I have disabled write protect via the screw-head, which is still in. Could that have an effect?

none at all

njalooo, 2024-10-29:
I'm guessing that there might be some powersaving feature, that should turn off USB power while the Computer is turned off or something.
Is there anything else i can do to help debug the problem?

cielmera, 2024-11-01:
Is it possible to use the firmware version before the latest ?, because i think in the earlier version there are people who got all the usb working

njalooo, 2024-11-01:
Thank you a lot, very much appreciated :)
I can test the changes by Monday, and i will let you know how it goes.

MrChromebox, 2024-11-01:
fix pushed - please update and LMK if the issue has been resolved

cielmera, 2024-11-02:
I will test it at sunday, currently out of city, thanks for the support and immediate response

njalooo, 2024-11-04:
Hey, your fix works great, the USB ports are working now, thanks a lot. :)
Final Debug Log: 
[debug-logs-Karma-2024-11-04_12h58m.tar.gz](https://github.com/user-attachments/files/17617911/debug-logs-Karma-2024-11-04_12h58m.tar.gz)

For anyone else with the same device, here are the roadblocks i encountered:
There is a reset button/pinhole on the back, which you need to press with a 1-2 cm long pin. With it it's possible to enter recover/developer mode. No need to press any keys.
To open the device, you need to remove the screws around the monitor stand, by popping of the plastic, as well as 5 screws which are hidden under a bezel on the front. To remove, pull while pulling it downwards. Then there are more plastic clips around the edge, you need to ply it apart.
Audio doesn't work out of the box with the 6.1.0-26-amd64 Linux Kernel. I had to pass the 
`options snd-intel-dspcfg dsp_driver=1`  kernel parameter in an /etc/modprobe.d/ file, to get hdmi audio working. Internal speakers still don't work for me, but that's fine for me.
Other than that everything seems to work just fine. Bluetooth, WiFi, Ethernet, USB, Touchscreen.

closed 2024-11-04 by njalooo

MrChromebox, 2024-11-04:
@njalooo thanks for confirming the fix.

internal speakers should work with the chrultrabook audio setup script:
https://docs.chrultrabook.com/docs/installing/installing-linux.html#fixing-audio

if not, post on their forums for help as the maintainer is active there

cielmera, 2024-11-04:
Thanks a lot. I also confirmed it works wonder on windows, the minus is only the audio not working with normal driver, i guess i need to buy the coolstar driver