MrChromebox/firmware #694: CN60 after UEFI bios installed won't power up

squoyster, 2024-10-29:
I went through the same process on two CN60s.  Installed full UEFI.  After reboot into initial logo screen, I shutdown the CN60 while in the boot logo screen.  Then I upgraded the SSD.  When I went back to powerup, nothing.  Light doesn't light up.  Seems like this bricked it or I must have done something to it while replacing the SSD (though I don't think so.)

1. Tried various key combinations
2. Reset + power
3. Various combinations of these while inserting the power plug
4. Took it apart and removed the CMOS battery for a while.

I verified that the board is getting +5 to the board, though I didn't check everything exhaustively.
I tried with original SSD, etc.

For the second machine, I waited until the boot up finished then rebooted into the UEFI menu before shutting down and installing new SSD.  All worked fine.

Is it possible that early shutdown bricked it?  Any other ideas for bringing it back to life?

I may try a rewrite of the EEPROM with the CH314a if something there got borked.  

TIA

MrChromebox, 2024-10-29:
> Is it possible that early shutdown bricked it? 

almost zero

> Any other ideas for bringing it back to life?

since you said it does not power on at all (no light), I suspect you dislodged the cable next to the SSD which connects the power button/LED to the mainboard.

 > I may try a rewrite of the EEPROM with the CH314a if something there got borked.

I highly suspect the firmware is not the problem here. If the device were bricked, you'd still get at least a flashing, if not solid power LED

squoyster, 2024-10-31:
The cable is firmly connected on the motherboard side, so unless it can be
dislodged on the switch side, it doesn't seem that it would be a problem
unless
I managed to break a wire.  However, I'll focus my attention there.

Thank you for your reply.  It helps me to know where to focus.