MrChromebox/firmware #696: Dell Latitude 5430

Abhayraj1984, 2024-11-01:
hi , i have Dell Latitude 5430 chrome book , 
as per i have followed all the steps to insatll windows in it 
as per the script i have downloaded the ufi rom 
and got insatlled also , after all the steps it asked for the restart ,
once it got restarted , the system is not booting , 
i am not getting the dispaly only 
only the laptop is getting turned on , but there is no display 
can i get any help regarging it

alexinua, 2024-11-02:
Hi, I also have one dell 5430 with no display after the full UEFI bios loading. It was suggested to wait for the first boot but it never happened. Before UEFI I have loaded linux/chromebook bios and it was working fine with it, only can not install ubuntu 22 on it, so decided to install full UEFI version. Unbricking with cable also doesn't work, device has chip W25Q256JV_M and flashrom is not capable to write original bios, it is failing on verification. Can you please suggest how to unbrick ?

MrChromebox, 2024-11-02:
>  Unbricking with cable also doesn't work, device has chip W25Q256JV_M and flashrom is not capable to write original bios, it is failing on verification. Can you please suggest how to unbrick ?

please run flashrom with the `-o flashrom.log` param and attach the file here (or elsewhere and link it here).
ccd is opened when flashing?

alexinua, 2024-11-03:
Thank you for fast reply. CCD was opened. I think my cable has a problems, I can read and write data using it and no complains, but 2 consecutive reads gives me different results. I compared 5 readings results and all of them are different. Diff shows many places are not same. I am going to get another cable.

alexinua, 2024-11-06:
Hi, I have original cable, still can not write flash, attaching a log file.
[flashrom.log](https://github.com/user-attachments/files/17640276/flashrom.log)

MrChromebox, 2024-11-06:
either:
1) flashrom is reading garbage
2) the IFD is actually garbage
3) flashrom is reading fine, but can't parse the IFD

do a read of the whole chip (ie, without `--ifd -i bios`) then zip and attach it here (assuming the read succeeds - if not, then the log would be useful)

alexinua, 2024-11-07:
Ok, reading the whole chip:
[new_read.zip](https://github.com/user-attachments/files/17654814/new_read.zip)
[reading.log](https://github.com/user-attachments/files/17654817/reading.log)

MrChromebox, 2024-11-07:
that read is all garbage. I'm not sure if your SuzyQ is reading properly, but if writing the entire firmware image fails, then I suspect there is some problem with the CCD reading/writing

MrChromebox, 2025-01-05:
I've figured out the issue and will be posting an updated build for CROTA shortly

alexinua, 2025-01-05:
Thank you for the followup. I did restore macbook last week. I have fix problem with inconstant ROM reading by disconnecting  akku and by adding an extra parameter "raiden_debug_spi:target=AP,custom_rst=True" to read command. Without  "custom_rst=True" it was restarting at the end of reading and next reading result was different. After I've got consistent reading I flashed my backup bios  - no result. Then I have got a full bios from the similar device, changed IDs and flashed it - no result. The last resort was to download a shellball bios and it did work. My chromebook now always in dev mode, but I didn't follow up further with recovering those setting.  Successfully installed RW legacy and Ubuntu 24 LTS (external sound, but USBC are working). Let me know if you have any further questions or like to test something.

MrChromebox, 2025-01-06:
I've just pushed fixed builds (built from the branch for the upcoming 2412 release) for CROTA/CROTA360. They can be manually downloaded at:

https://mrchromebox.tech/files/firmware/full_rom/coreboot_edk2-crota-mrchromebox-20250105.rom
https://mrchromebox.tech/files/firmware/full_rom/coreboot_edk2-crota-mrchromebox-20250105.rom.sha1
https://mrchromebox.tech/files/firmware/full_rom/coreboot_edk2-crota360-mrchromebox-20250105.rom
https://mrchromebox.tech/files/firmware/full_rom/coreboot_edk2-crota360-mrchromebox-20250105.rom.sha1

the firmware util script has been update to use these as well.

SameerKP1999, 2025-01-07:
> use

So can i flash this using suzyqable

SameerKP1999, 2025-01-07:
> hi , i have Dell Latitude 5430 chrome book ,
> as per i have followed all the steps to insatll windows in it
> as per the script i have downloaded the ufi rom
> and got insatlled also , after all the [...]

Bro can you share the firmware backup

SameerKP1999, 2025-01-07:
> I've just pushed fixed builds (built from the branch for the upcoming 2412 release) for CROTA/CROTA360. They can be manually downloaded at: [...]

Can't manually download using this link. Pls check

MrChromebox, 2025-01-07:
> Can't manually download using this link. Pls check

typo, fixed in original post above

SameerKP1999, 2025-01-07:
> > Can't manually download using this link. Pls check
> 
> typo, fixed in original post above

Still it is showing not found

MrChromebox, 2025-01-07:
> Still it is showing not found. Pls check the link using another device.

no, it's not, you click the link and the file downloads. All 4 files are good.

MrChromebox, 2025-08-07:
ok, well I'll have to create a debug build with serial logging next week when I'm back home, and have you flash that / send me the log. Feel free to email me directly or ping me on Discord/IRC etc

geld1an, 2025-08-15:
Hi guys. I also have a Dell 5430 Chromebook with MrChromebox firmware, I bought it with this firmware. I spent a lot of time installing sound drivers and other stuff on Windows 10/11, but nothing helped. Now I need the original firmware to install ChromeOS (by the way, developer mode does not work). Maybe you can help me?