MrChromebox/firmware #697: NVMe drive not recognized after UEFI firmware installation

0neShot101, 2024-11-12:
Hello, I recently installed the UEFI firmware on my Lenovo Flex i5 Chromebook Plus. Since then, the stock NVMe drive that came with the Chromebook is no longer detected in the UEFI environment or by Linux installers. It worked fine in ChromeOS before installing the UEFI firmware, so the drive should be functional.

Is this a known issue? Are there any solutions or workarounds to get the stock NVMe drive recognized? Or would it be more straightforward to replace the drive with a new NVMe compatible with this setup?

Thanks for any help or guidance you can provide.

0neShot101, 2024-11-12:
Board Name: Taeko
cbmem log: [cbmem_log.txt](https://github.com/user-attachments/files/17711221/cbmem_log.txt)

MrChromebox, 2024-11-13:
this looks like it's the same issue as #680, so let's close this and keep everything under one issue

closed 2024-11-13 by MrChromebox