MrChromebox/firmware #698: Chromebook Acer CB 315 3H doesn’t run the script

TheWildDino, 2024-11-12:
When I ran the script of the dev mode 2(ctrl alt right arrow after getting into normal dev mode) the result of entering the script was curl option-LOmrchromebox.tech/firmware-until.sh: expected. a proper numerical parameter 

I did disable the WP but disconnecting the battery.

TheWildDino, 2024-11-13:
[image]

I was watching a tutorial and it only entered the curl command so that’s what I did then I went to the website and saw it included sudo so that’s the second one I ran.

TheWildDino, 2024-11-13:
Here’s the image 
[image]

TheWildDino, 2024-11-13:
[image]

TheWildDino, 2024-11-13:
[image]

MrChromebox, 2024-11-13:
I don't want this to sound mean, but if you're not able to type the commands in exactly as listed in my documentation, you're going to have a really hard time with the rest of the process. Please read the documentation on my site and try to understand what you're doing, so you're not blindly copying things (incorrectly) without any idea what they are doing

closed 2024-11-13 by MrChromebox