MrChromebox/firmware #699: Can't install Install/Update (Full Rom) Firmware on my Dell Chromebook 11 3189

amdiebs, 2024-11-12:
Hello, Mr.
I would like to reinstall my laptop's operating system from ChromeOS to Windows 10. Unfortunately, an error '(see screenshot) appeared during the installation of the Full ROM firmware after all the preliminary steps had been performed (remove screws, connect in developer mode) before, and it was not possible to continue the procedure.
I need help with this.
[image]

PS: file flashrom.log is joint for more details. Thanks
[flashrom.log](https://github.com/user-attachments/files/17716206/flashrom.log)

MrChromebox, 2024-11-12:
it's not clear why the flash failed, I would restore the stock firmware manually, reboot, and then try flashing again.

restore manually:
`sudo flashrom -p internal -w /tmp/bios.bin --ifd -i bios -N`

amdiebs, 2024-11-13:
I didn't mention this but I did reboot after the restore before trying flashing again.
I did it again to reassure you, but same error. 
just feel like crying...

MrChromebox, 2024-11-13:
> just feel like crying...

because your Chromebook is having an issue flashing the firmware?

amdiebs, 2024-11-13:
Yes, because of that, it was supposed to be a formality. But ultimately time-consuming.

MrChromebox, 2024-11-13:
please provide the output of `sudo flashrom -p internal --wp-status`

amdiebs, 2024-11-13:
[image]

amdiebs, 2024-11-15:
you saw finally what's wrong?

MrChromebox, 2024-11-15:
no, otherwise I would have replied. I'm going to go back to stock on one of my Braswell devices and try to reproduce this issue

foxboy2005, 2024-12-28:
I am also having the same issue. I am hoping for the solution ASAP. Thank you Mrchromebox!

MrChromebox, 2024-12-28:
I just reset my EDGAR board back to stock and flashed via the script and no issue at all. Checked the flashrom log and no issues there. Comparing the log provided by the OP, it seems they are using an older version of ChromeOS/older version of flashrom. Also, the boards use a different flash chip

closed 2026-11-04 by MrChromebox