MrChromebox/firmware #700: USBC Ports on Voxel TGL Only Work for Power, no Data Transfer

rainingcrypto, 2024-11-24:
As the title suggest, USBC Ports on Voxel TGL Only Work for Power, no Data Transfer:
Running Fedora using the latest firmware, and the 2 USBC ports only work for power/charging, no data transfer. Tried installing various drivers with no success. I know you are aware of this, as your latest firmware updates mention "added more USBs for TGL chromebooks". There is no rush to fix this, just raising so it's on your radar. 

WE LOVE YOU MR. CHROMEBOX, thank you. <3

MrChromebox, 2024-11-25:
I'm not aware of any outstanding issues with USB-C ports on TGL ChromeOS devices. TB ports may require some OS specific enablement - see https://docs.chrultrabook.com

MrChromebox, 2025-12-03:
closing as this was an OS issue and likely resolved by now