MrChromebox/firmware #704: Something went wrong launching the alternate bootloader, edk2 / Redrix

mikiadev, 2024-12-07:
I used firmware utility script on my Redrix (HP Dragonfly Elite) following the instructions and installed/updated RW_LEGACY (in an attempt to dual boot), the script runs fine but when selecting from Developer Mode screen Select alternate boot loader it gives edk2 and selecting it gives "Something went wrong launching the alternate bootloader. View firmware log for details.", I tested several times rebooting and reinstalling, also with battery removed but no difference.

Any idea what could be the issue? Something from the logs I can dig or is my only option going for the full UEFI rom?

kittyjosh111, 2024-12-25:
same issue, also on google-redrix.
As stated above, the script seems to work correctly, and the option to launch the alternate bootloader does show up in the developer mode screen on the next boot. However, selecting the edk2 option or pressing ctrl+L both result in the error message reported.

MrChromebox, 2024-12-26:
I suspect Google changed something related to the launching of the AltFw payloads on the Alderlake boards, as they were all working. I don't have one here to test, but think I have an idea what the problem is

newinnov, 2025-02-23:
@MrChromebox I am also facing the same issue.
Newer Acer CB515-2H (CB515-2H-34YL ) running on Raptor Lake (1315U), also has this problem. 
Though coolstar website says- it's supported, as last working modal has same board 'brya' and same product code 'omnigul', the only difference being 13th gen cpu instead of 12th gen and nvme drive.

MrChromebox, 2025-02-23:
> Though coolstar website says- it's supported

"supported" on coolstar's website means there is UEFI Full ROM firmware available for a device and Windows drivers; RW_LEGACY/AltFw does not come into play here. Firmware support is entirely handled by me - which is why you're opening this issue on my issue tracker and not coolstar's.

as above, Google seem to have broken AltFw on some Brya boards. My Banshee works fine, which makes it hard to troubleshoot. What are the RO and active firmware versions on your device, as shown on the debug info screen on the Developer Mode menu?

newinnov, 2025-02-24:
Hi, Thanks for responding. 
Active firmware: Google_Omnigul.14505.769.0
ro firmware: Google_Omnigul.14505.635.0

> UEFI Full ROM firmware available for a device`
Does that mean - even if RW_LEGACY/AltFw is not currently not working for this device, Flashing UEFI full rom would be fine and it will work?

mikiadev, 2025-02-24:
@newinnov flashing UEFI full rom likely would be fine, when I couldn't get legacy working on my Redrix I decided to go for UEFI and it works without any issues.

newinnov, 2025-02-24:
Thanks @mikiadev for confirmation, as my device is new and I do not have SuzyQ... I'm a little bit worried about doing full rom.

closed 2025-12-03 by MrChromebox