MrChromebox/firmware #706: Sleep / hibernation failure on FIZZ / TEEMO / KENCH (ASUS CN65 / HP G2) units

dan751live, 2024-12-08:
It seems the Quanta motherboard FIZZ based units stop responding, requiring a forced power off, when power management settings are enabled.

Tested with latest version [MrChromebox-2408.1] (2024.09.14) firmware on Linux Mint 22 and Windows 11 24H2.

Sleep features work fine with ChromeOS Flex.

Sleep (turn off my screen and / or put device to sleep) results in system locking up, cannot be restored using keyboard / mouse / power button.

Hibernate works, but often loses the hibernation data and just restarts (fresh boot state).

This makes the units only suitable as a server or Kodi home theatre, but not for desktop PC use, unless I'm missing something.

I have a few of these units (HP G2), so if one is required for testing, I can mail it to the developer in USA / Canada free of charge.

P.S. other less critical isssues observed:

1. No TPM module detected, even though the CPU can support it ( via fTPM? )

2. Linux & Windows both report the device has a battery charged at 100%, which should not apply to these Chromeboxes

MrChromebox, 2024-12-08:
I'm unable to replicate any suspend issues here on my Fizz running 2408.1 firmware and Pop_OS 22.04/Win11 24H2 -- both sleep and resume properly. Hibernate is not supported.

1) Google's CR50 TPM is not a full TPM 2.0 implementation and cannot be used as a TPM by Windows

2) Google's EC reports the battery isn't present. Where are you seeing the battery is 100%? I do not have a battery indicator on the taskbar for Windows or Linux.

syscl, 2025-01-28:
Hi @MrChromebox thanks for providing such a great firmware for cn65 (fizz). I unfortunately hitting the same issue on ubuntu 24.04 and 24.04.1 LTS. Waking from the system seems to have no response (a plain screen), any logs/info I can provide for debugging? 

Thanks in advance,
syscl

MrChromebox, 2025-01-31:
can you boot a 24.04.1 LTS live USB and check if reproducible there? I'm not able to reproduce it here on multiple Fizz i7-8550U boxes. Both s2idle and deep sleep work properly.

syscl, 2025-02-01:
Thanks @MrChromebox , I'm currently away from the box, however, I did try with Fedora and wake from sleep work properly. I think it is pretty likely a system related issue e.g. Ubuntun/debian, I will let you know the result once back to the box. All the best!