MrChromebox/firmware #708: Possible to get Serial Console redirection via Suzy Q cable?

afustinoni-mdsol, 2024-12-12:
I have a couple of Asus CN65's and a CN62 that I use as Proxmox hosts for messing around.

They tiny and can be stuffed in out-of-the-way places, but when for some reason I need to reinstall the OS, plugging in a monitor, mouse and keyboard is a massive pain.

Is there a way to get serial console redirection (ideally for the UEFI menu as well as bootloader + Linux console out, but even just Bootloader and after would be a massive win) with a Suzy Q cable?

MrChromebox, 2024-12-12:
the CN65 supports SuzyQ but the CN62/60 do not. You'd need a serial console enabled build, and probably early console in your kernel - no sure on that part exactly.

ellyq, 2025-05-22:
For future reference: Most Intel-based devices route LPSS UART to CR50, so you can get serial console output by adding `console=ttyS4,115200n8` to kernel commandline.