MrChromebox/firmware #711: HP Chromebook 11 G4 black screen after uefi flash mrchomebox

japersonpadin, 2025-01-09:
Only black screen

* I already do unbricking guide in your website using ch341a it succes flash but still black screen nu luck can you help me ❤️
[image]
[image]
[image]

MrChromebox, 2025-01-09:
this is a KIP device? just flash the latest UEFI image

japersonpadin, 2025-01-09:
> this is a KIP device? just flash the latest UEFI image

How 🥹

MrChromebox, 2025-01-09:
re-flash `coreboot_edk2-kip-mrchromebox_20240914.rom`

https://mrchromebox.tech/files/firmware/full_rom/coreboot_edk2-kip-mrchromebox_20240914.rom

MrChromebox, 2025-01-10:
the exact same way you flashed before, using a ch341a USB programmer.

the guide is on my site.

japersonpadin, 2025-01-10:
ok thank you 😊 i hope it will work

japersonpadin, 2025-01-10:
[image]
[image]
[image]
Still no luck 🥲 no display

MrChromebox, 2025-01-10:
then I suspect something else is wrong with the hardware, if neither the stock backup nor the UEFI image will boot on the device

YTEC-info, 2025-01-10:
>       ![IMG_3011](https://private-user-images.githubusercontent.com/41370359/401584891-2f6de8f6-4270-4938-baf0-4a4c3047312d.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY1MTc0MjUsIm5iZiI6MTczNjUxNzEyNSwicGF0aCI6Ii80MTM3MDM1OS80MDE1ODQ4OTEtMmY2ZGU4ZjYtNDI3MC00OTM4LWJhZjAtNGE0YzMwNDczMTJkLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDExMFQxMzUyMDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hODVkZjMwYjg5MTFiMjliOWVhNDNiYzNiZGI4YzY1MDY3ZGM5MzNjYmQ0NTAzNzAxOGQ3NDA0ODA2ZmY2ODAxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.5ULpmFk5QJA88HQJP8SZW3A35_S1DLPuS7NfWkfmWvM)

I hope there is some protection underneath the board to prevent a short circuit with the computer 🤯

japersonpadin, 2025-01-10:
> I hope there is some protection underneath the board to prevent a short circuit with the computer 🤯

Yes it have underneath protection plastic coated

japersonpadin, 2025-01-10:
> then I suspect something else is wrong with the hardware, if neither the stock backup nor the UEFI image will boot on the device

Yes i think so, my #1 suspect is emmc
Is it possible to do flashing  or write / erase on emmc via isp pin out ?

japersonpadin, 2025-01-10:
Im wondering when the device is power on there are something black dim in screen and disappeared immediately, i think  device is booting but due to compatibility issue the result is no display output . Correct if im wrong 🙂

MrChromebox, 2025-01-10:
if the display backlight is coming on, that means it's booting to a certain point at least. It's possible that some KIP boards don't like the edk2 GOP driver init, so I'll build you one with the older legacy VBIOS init to try

japersonpadin, 2025-01-11:
No luck 🥲

> try this image with VGA BIOS init and see if it works for you
> 
> [coreboot_edk2-kip-mrchromebox_20250110.zip](https://github.com/user-attachments/files/18380072/coreboot_edk2-kip-mrchromebox_20250110.zip)

No luck 🥲

MrChromebox, 2025-01-11:
does the screen backlight come on with that test image?

japersonpadin, 2025-01-23:
- 

> does the screen backlight come on with that test image?

no dim only completely black, now the status is totaly dead i already parts all components and transplant

MrChromebox, 2025-01-23:
very strange. can you re-flash the stock firmware?

closed 2026-11-04 by MrChromebox